open-mmlab/OpenPCDet #1525: spconv1.2.1没有core目录，导致运行demo.py时报错

swaggywilliam, 2023-12-06:
我按照[spconv1.x](https://github.com/traveller59/spconv/tree/v1.2.1)的Install on Ubuntu 16.04/18.04教程，从源码安装的spconv1.2.1，结果就是OpenPCDet的demo无法正常运行。
以下是环境依赖：
```
Package             Version       Editable project location
------------------- ------------- ----------------------------------
apptools            5.2.1
blessed             1.20.0
ccimport            0.4.2
certifi             2023.11.17
charset-normalizer  3.3.2
configobj           5.0.8
contourpy           1.1.1
cumm                0.4.11
cycler              0.12.1
easydict            1.11
envisage            7.0.3
fire                0.5.0
fonttools           4.46.0
gpustat             1.1
idna                3.6
imageio             2.33.0
importlib-metadata  7.0.0
importlib-resources 6.1.1
kiwisolver          1.4.5
lark                1.1.8
lazy_loader         0.3
llvmlite            0.41.1
matplotlib          3.7.4
mayavi              4.8.1
networkx            3.1
ninja               1.11.1.1
numba               0.58.1
numpy               1.24.4
nvidia-ml-py        11.525.112
opencv-python       4.8.1.78
packaging           23.2
pccm                0.4.11
pcdet               0.6.0+255db8f /home/zhuweilian/project/OpenPCDet
Pillow              10.1.0
pip                 23.3.1
portalocker         2.8.2
protobuf            4.25.1
psutil              5.9.5
pybind11            2.11.1
pyface              8.0.0
Pygments            2.17.2
pyparsing           3.1.1
pyquaternion        0.9.9
python-dateutil     2.8.2
PyWavelets          1.4.1
PyYAML              6.0.1
requests            2.31.0
scikit-image        0.21.0
scipy               1.10.1
setuptools          68.0.0
SharedArray         3.2.3
six                 1.16.0
spconv              1.2.1
tensorboardX        2.6.2.2
termcolor           2.4.0
tifffile            2023.7.10
torch               1.8.1+cu111
torchaudio          0.8.1
torchvision         0.9.1+cu111
tqdm                4.66.1
traits              6.4.3
traitsui            8.0.0
typing_extensions   4.8.0
urllib3             2.1.0
vtk                 9.3.0
wcwidth             0.2.6
wheel               0.41.2
zipp                3.17.0
```
[image]
[image]

当我运行OpenPCDet的demo时，使用指令：
```
python demo.py --cfg_file /home/project/OpenPCDet/tools/cfgs/kitti_models/voxel_rcnn_car_focal_multimodal.yaml --data_path /public/kitti/training/velodyne_reduced/000008.bin --ckpt /home/project/ckpt/voxel_rcnn_car_84.54.pth
```
报错：
```
AttributeError: module 'spconv' has no attribute 'SparseModule'
```
查阅issues，出现这个问题的一般都是spconv2.x，可我是spcon1.2.1，为什么还会出错？
参考spconv2.x的解决方式，于是我将import spconv修改为import spconv.pytorch as spconv，运行demo，还是报错：
```
ModuleNotFoundError: No module named 'spconv.core_cc'
```
通过对比spcon2.x的安装路径，确实存在着core这个目录，而通过源码安装的spconv1.2.1却没有目录core，于是我又尝试着将spconv2.x的core目录复制到了spconv1.2.1的路径下，但又报错：
```
ImportError: cannot import name 'SpconvOps' from 'spconv.core_cc.csrc.sparse.all' (unknown location)
```
然而复制过来的core目录下是包含着csrc.sparse.all的，不明白为什么会报错unknown location。
我想知道到底该如何“正确安装spconv1.2.1”呢？请帮助我，谢谢！

751K, 2023-12-10:
也许试着安装第二个spconv v2.x .

751K, 2023-12-10:
Try to install v2.x seconv .It’s useful for me.My environment is Cuba:11.8,PyTorch 2.13

swaggywilliam, 2023-12-10:
> Try to install v2.x seconv .It’s useful for me.My environment is Cuba:11.8,PyTorch 2.13

yeah, I installed v2.1 and don't met 'SparseModule' and 'spconv.core_cc' any more, But some new error arose when I ran the test.py:
```
sample_idx = info['point_cloud']['lidar_idx']  
TypeError: string indices must be integers
```
I am trying to solve it now...

751K, 2023-12-11:
> yeah, I installed v2.1 and don't met 'SparseModule' and 'spconv.core_cc' any more, But some new error arose when I ran the test.py:
>
> ```
> sample_idx = info['point_cloud']['lidar_idx']
> TypeError: [...]

This problem has nothing to do with spconv.You ‘d better to check whether you info.pkl is right.

swaggywilliam, 2023-12-11:
> This problem has nothing to do with spconv.You ‘d better to check whether you info.pkl is right.

Thank you! You are right, I solved it shortly after replying to you.

closed 2023-12-12 by swaggywilliam