open-mmlab/OpenPCDet #1527: RuntimeError: The size of tensor a (16) must match the size of tensor b (4) at non-singleton dimension 4

swaggywilliam, 2023-12-12:
When I try to train the [model](https://github.com/LittlePey/SFD) which based on openpcdet,
```
python train.py --cfg_file cfgs/kitti_models/sfd.yaml --batch_size 2 --gpu_id 0 --workers 0 
```
error happened
```
2023-12-11 13:11:46,263   INFO  **********************Start training kitti_models/sfd(default)**********************
epochs:   0%|                                                                                                                                           | 0/6 [00:00<?, ?it/sBefore encoding pseudo: (275205, 9)                                                                                                                   | 0/3712 [00:00<?, ?it/s]
After encoding pseudo: (275205, 9)
epochs:   0%|                                                                                                                                           | 0/6 [00:21<?, ?it/s]
Traceback (most recent call last):
  File "/home/project/SFD/tools/train.py", line 203, in <module>
    main()
  File "/home/project/SFD/tools/train.py", line 158, in main
    train_model(
  File "/home/project/SFD/tools/train_utils/train_utils.py", line 86, in train_model
    accumulated_iter = train_one_epoch(
  File "/home/project/SFD/tools/train_utils/train_utils.py", line 42, in train_one_epoch
    optimizer.step()
  File "/home/project/SFD/tools/train_utils/optimization/fastai_optim.py", line 149, in step
    self.opt.step()
  File "/home/anaconda3/envs/sfd/lib/python3.9/site-packages/torch/optim/optimizer.py", line 89, in wrapper
    return func(*args, **kwargs)
  File "/home/anaconda3/envs/sfd/lib/python3.9/site-packages/torch/autograd/grad_mode.py", line 27, in decorate_context
    return func(*args, **kwargs)
  File "/home/anaconda3/envs/sfd/lib/python3.9/site-packages/torch/optim/adam.py", line 108, in step
    F.adam(params_with_grad,
  File "/home/anaconda3/envs/sfd/lib/python3.9/site-packages/torch/optim/_functional.py", line 84, in adam
    exp_avg.mul_(beta1).add_(grad, alpha=1 - beta1) # exp_avg * beta1 + grad * (1 - beta1)
RuntimeError: The size of tensor a (16) must match the size of tensor b (4) at non-singleton dimension 4
```
and I found that

**grad shape: torch.Size([16, 3, 3, 3, 4]) 
exp_avg shape: torch.Size([3, 3, 3, 4, 16])**

Why are the shapes of these two tensors different, the shape of the tensor becomes like this through the optimizer, and I don't know where to start troubleshooting, could you please help me analyze the reason for this issue? Is it an error in my dataset?

closed 2024-01-06 by swaggywilliam

AJAY31797, 2024-01-20:
Hey @swaggywilliam, 

I got a similar error while training my custom dataset. 
`The size of tensor a (155531250) must match the size of tensor b (261000000) at non-singleton dimension 1
  File "/home/aagr656/OpenPCDet/pcdet/utils/loss_utils.py", line 60, in forward
    pt = target * (1.0 - pred_sigmoid) + (1.0 - target) * pred_sigmoid
  File "/home/aagr656/OpenPCDet/pcdet/models/dense_heads/anchor_head_template.py", line 128, in get_cls_layer_loss
    cls_loss_src = self.cls_loss_func(cls_preds, one_hot_targets, weights=cls_weights)  # [N, M]
  File "/home/aagr656/OpenPCDet/pcdet/models/dense_heads/anchor_head_template.py", line 217, in get_loss
    cls_loss, tb_dict = self.get_cls_layer_loss()
  File "/home/aagr656/OpenPCDet/pcdet/models/detectors/pointpillar.py", line 27, in get_training_loss
    loss_rpn, tb_dict = self.dense_head.get_loss()
  File "/home/aagr656/OpenPCDet/pcdet/models/detectors/pointpillar.py", line 14, in forward
    loss, tb_dict, disp_dict = self.get_training_loss()
  File "/home/aagr656/OpenPCDet/pcdet/models/__init__.py", line 44, in model_func
    ret_dict, tb_dict, disp_dict = model(batch_dict)
  File "/home/aagr656/OpenPCDet/tools/train_utils/train_utils.py", line 56, in train_one_epoch
    loss, tb_dict, disp_dict = model_func(model, batch)
  File "/home/aagr656/OpenPCDet/tools/train_utils/train_utils.py", line 180, in train_model
    accumulated_iter = train_one_epoch(
  File "/home/aagr656/OpenPCDet/tools/train.py", line 176, in main
    train_model(
  File "/home/aagr656/OpenPCDet/tools/train.py", line 231, in <module>
    main()
RuntimeError: The size of tensor a (155531250) must match the size of tensor b (261000000) at non-singleton dimension 1`

How did you solve it?

Thanks.

swaggywilliam, 2024-01-23:
> Hey @swaggywilliam,
>
> I got a similar error while training my custom dataset. `The size of tensor a (155531250) must match the size of tensor b (261000000) at non-singleton dimension 1 File [...]

The error I met was because I didn't modify the '_BASE_CONFIG_' in my 'model.yaml', hope it can be helpful to you.