open-mmlab/OpenPCDet #1538: Error in collate_batch: key=img_process_infos (nuScenes dataset)

fang196, 2024-01-05:
When evaluating bevfusion on nuScenes dataset, this error occur in pcdet/datasets/dataset.py-->collate_batch.

In pcdet/datasets/nuscenes/nuscenes_dataset.py line 151, `img_process_infos` is stored in `input_dict`, but can not be processed by `collate_batch` function.
```python
input_dict['img_process_infos'] = img_process_infos
input_dict['camera_imgs'] = crop_images
```

fang196, 2024-01-05:
setting an array element with a sequence. The requested array has an inhomogeneous shape after 2 dimensions. The detected shape was (6, 4) + inhomogeneous part.
  File "/home/qxy/pcdet-clip/pcdet/datasets/dataset.py", line 319, in collate_batch
    ret[key] = np.stack(val, axis=0)
ValueError: setting an array element with a sequence. The requested array has an inhomogeneous shape after 2 dimensions. The detected shape was (6, 4) + inhomogeneous part.

During handling of the above exception, another exception occurred:

  File "pcdet/pcdet/datasets/dataset.py", line 322, in collate_batch
    raise TypeError
  File "/pytorch/torch/utils/data/_utils/fetch.py", line 47, in fetch
    return self.collate_fn(data)
  File "/torch/utils/data/dataloader.py", line 561, in _next_data
    data = self._dataset_fetcher.fetch(index)  # may raise StopIteration
  File "/torch/utils/data/dataloader.py", line 521, in __next__
    data = self._next_data()
  File "pcdet/tools/eval_utils/eval_utils.py", line 58, in eval_one_epoch
    for i, batch_dict in enumerate(dataloader):
  File "pcdet/tools/test_bevfusion_debug.py", line 72, in eval_single_ckpt
    eval_utils.eval_one_epoch(
  File "pcdet/tools/test_bevfusion_debug.py", line 210, in main
    eval_single_ckpt(model, test_loader, args, eval_output_dir, logger, epoch_id, dist_test=dist_test)
  File "pcdet/tools/test_bevfusion_debug.py", line 214, in <module>
    main()

closed 2024-02-20 by github-actions[bot]

DeFisch, 2024-02-20:
Same issue here when running bevfusion, any insights?

CaptainRoby, 2025-03-06:
i solve this problem by adding
"elif key in ['img_process_infos']:
                    dt = np.dtype([
                        ('score', float),
                        ('bbox', (int, 4)),
                        ('flag', bool),
                        ('class_id', int)
                    ])
                    structured_val = [np.array(sublist, dtype=dt) for sublist in val]
                    ret[key] = np.stack(structured_val, axis=0)"
before the else block.

fuzio35, 2025-03-26:
> i solve this problem by adding "elif key in ['img_process_infos']: dt = np.dtype([ ('score', float), ('bbox', (int, 4)), ('flag', bool), ('class_id', int) ]) structured_val = [np.array(sublist, dtype=dt) for sublist in val] ret[key] = np.stack(structured_val, axis=0)" before the else block.

hello bro！i want to know which file did you change? thanks！

fuzio35, 2025-03-26:
> i solve this problem by adding "elif key in ['img_process_infos']: dt = np.dtype([ ('score', float), ('bbox', (int, 4)), ('flag', bool), ('class_id', int) ]) structured_val = [np.array(sublist, dtype=dt) for sublist in val] ret[key] = np.stack(structured_val, axis=0)" before the else block.

Sincerely thank you for your solution! It works to me!