open-mmlab/OpenPCDet #1545: Error while creating data infos using custom dataset

AJAY31797, 2024-01-15:
Hi, 

I am trying to train and test my custom point cloud dataset. As per the steps provided in CUSTOM_DATASET_TUTORIAL.md, I created labels, file structure, hyper-parameters configurations and made changes in category names and anchor sizes. On running the create data info using the following command
`python -m pcdet.datasets.custom.custom_dataset create_custom_infos tools/cfgs/dataset_configs/custom_dataset.yaml`
I am seeing the following error. 
`Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 187, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.10/runpy.py", line 110, in _get_module_details
    __import__(pkg_name)
  File "/home/ajay/OpenPCDet/pcdet/datasets/__init__.py", line 15, in <module>
    from .argo2.argo2_dataset import Argo2Dataset
  File "/home/ajay/OpenPCDet/pcdet/datasets/argo2/argo2_dataset.py", line 15, in <module>
    from .argo2_utils.so3 import yaw_to_quat, quat_to_yaw
  File "/home/ajay/OpenPCDet/pcdet/datasets/argo2/argo2_utils/so3.py", line 10, in <module>
    def quat_to_mat(quat_wxyz: Tensor) -> Tensor:
  File "/home/ajay/.local/lib/python3.10/site-packages/torch/jit/_script.py", line 1381, in script
    fn = torch._C._jit_script_compile(
  File "/home/ajay/.local/lib/python3.10/site-packages/torch/jit/_recursive.py", line 1010, in try_compile_fn
    return torch.jit.script(fn, _rcb=rcb)
  File "/home/ajay/.local/lib/python3.10/site-packages/torch/jit/_script.py", line 1381, in script
    fn = torch._C._jit_script_compile(
RuntimeError:
cannot statically infer the expected size of a list in this context:
  File "/home/ajay/.local/lib/python3.10/site-packages/kornia/geometry/conversions.py", line 556

    # this slightly awkward construction of the output shape is to satisfy torchscript
    output_shape = [*list(quaternion.shape[:-1]), 3, 3]
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~ <--- HERE
    matrix = matrix_flat.reshape(output_shape)
'quaternion_to_rotation_matrix' is being compiled since it was called from 'quat_to_mat'
  File "/home/ajay/OpenPCDet/pcdet/datasets/argo2/argo2_utils/so3.py", line 19
        (...,3,3) 3D rotation matrices.
    """
    return C.quaternion_to_rotation_matrix(
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        quat_wxyz, order=C.QuaternionCoeffOrder.WXYZ
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ <--- HERE
    )`

I am not sure why exactly this is happening. Could anyone suggest how to handle this error? I am using WSL2 to run the code. 
The Cuda version is 11.8, and the torch version and python version are compatible. 

Best Regards,
Ajay

closed 2024-01-17 by AJAY31797

Serzhanov, 2024-08-28:
@AJAY31797 Hey, I am encountered the same problem,how did you resolve it?

voodooed, 2024-12-03:
@AJAY31797 Can you please how did you handle this error? i am facing the same issue.