open-mmlab/OpenPCDet #1561: Custom dataset error KeyError: 'Bus'

Parkeunseok, 2024-02-13:
After turning the learning to Custom Datasets and turning the last epoch, the following error comes out. The classes I learned are Car, Bus, and Truck.
May I know how to solve it??

/home/a/deepnet/OpenPCDet/tools/../pcdet/datasets/kitti/kitti_object_eval_python/eval.py:10: NumbaDeprecationWarning: The 'nopython' keyword argument was not supplied to the 'numba.jit' decorator. The implicit default value for this argument is currently False, but it will be changed to True in Numba 0.59.0. See https://numba.readthedocs.io/en/stable/reference/deprecation.html#deprecation-of-object-mode-fall-back-behaviour-when-using-jit for details.
  def get_thresholds(scores: np.ndarray, num_gt, num_sample_pts=41):
Traceback (most recent call last):
  File "/home/a/deepnet/OpenPCDet/tools/train.py", line 230, in <module>
    main()
  File "/home/a/deepnet/OpenPCDet/tools/train.py", line 220, in main
    repeat_eval_ckpt(
  File "/home/a/deepnet/OpenPCDet/tools/test.py", line 123, in repeat_eval_ckpt
    tb_dict = eval_utils.eval_one_epoch(
  File "/home/a/deepnet/OpenPCDet/tools/eval_utils/eval_utils.py", line 125, in eval_one_epoch
    result_str, result_dict = dataset.evaluation(
  File "/home/a/deepnet/OpenPCDet/tools/../pcdet/datasets/custom/custom_dataset.py", line 136, in evaluation
    ap_result_str, ap_dict = kitti_eval(eval_det_annos, eval_gt_annos, self.map_class_to_kitti)
  File "/home/a/deepnet/OpenPCDet/tools/../pcdet/datasets/custom/custom_dataset.py", line 127, in kitti_eval
    ap_result_str, ap_dict = kitti_eval.get_official_eval_result(
  File "/home/a/deepnet/OpenPCDet/tools/../pcdet/datasets/kitti/kitti_object_eval_python/eval.py", line 662, in get_official_eval_result
    current_classes_int.append(name_to_class[curcls])
KeyError: 'Bus

J-xinyu, 2024-03-25:
> 将学习转向自定义数据集并转向最后一个纪元后，出现以下错误。我学的课程是汽车、公共汽车和卡车。 我可以知道如何解决吗？
> 
> /home/a/deepnet/OpenPCDet/tools/../pcdet/datasets/kitti/kitti_object_eval_python/eval.py:10：NumbaDeprecationWarning：未将“nopython”关键字参数提供给“numba.jit”装饰器。此参数的隐式默认值当前为 False，但在 Numba 0.59.0 中将更改为 True。有关详细信息，请参阅https://numba.readthedocs.io/en/stable/reference/deprecation.html#deprecation-of-object-mode-fall-back-behaviour-when-using-jit。 def get_thresholds(scores: np.ndarray, num_gt, num_sample_pts=41): 回溯（最近一次调用最后）： 文件“/home/a/deepnet/OpenPCDet/tools/train.py”，第 230 行，在 main() 文件中“/home/a/deepnet/OpenPCDet/tools/train.py”，第 220 行，在主 Repeat_eval_ckpt( 文件“/home/a/deepnet/OpenPCDet/tools/test.py”，第 123 行，在 Repeat_eval_ckpt tb_dict = eval_utils .eval_one_epoch( 文件 "/home/a/deepnet/OpenPCDet/tools/eval_utils/eval_utils.py"，第 125 行，在 eval_one_epoch result_str, result_dict = dataset.evaluation( 文件 "/home/a/deepnet/OpenPCDet/tools/. ./pcdet/datasets/custom/custom_dataset.py”，第 136 行，评估 ap_result_str, ap_dict = kitti_eval(eval_det_annos, eval_gt_annos, self.map_class_to_kitti) 文件“/home/a/deepnet/OpenPCDet/tools/../pcdet/ datasets/custom/custom_dataset.py"，第 127 行，在 kitti_eval ap_result_str, ap_dict = kitti_eval.get_official_eval_result( 文件 "/home/a/deepnet/OpenPCDet/tools/../pcdet/datasets/kitti/kitti_object_eval_python/eval.py" ，第 662 行，在 get_official_eval_result current_classes_int.append(name_to_class[curcls]) KeyError: 'Bus

Hey, I encountered the same problem, have you solved it now?

Parkeunseok, 2024-03-27:
> > 将school习转向自定义数据集并转向最后一个纪错误， 出现以下错误. 왜요?
> > /home/a/deepnet/OpenPCDet/tools/../pcdet/datasets/kitti/kitti_object_eval_python/eval.py:10：NumbaDeprecationWarning：未将“nopython”关键字参数提供给“numba.jit”装饰器。此参数的隐式默认值当前为 False，但下 Numba 0.59.0 中将更改为 True입니다 [. -object-mode-fall-back-behaviour-when-using-jit。](https://numba.readthedocs.io/en/stable/reference/deprecation.html#deprecation-of-object-mode-fall-back-behaviour-when-using-jit%E3%80%82) def get_thresholds(scores: np.ndarray, num_gt, num_sample_pts=41): 回溯（最近一次调用最后）： 文件“/home/a/deepnet/ OpenPCDet/tools/train.py”, 제 230행, main() 문서중“/home/a/deepnet/OpenPCDet/tools/train.py”, 제 220행, 주 Repeat_eval_ckpt( 문서“/home/a /deepnet/OpenPCDet/tools/test.py”, 제 123행, Repeat_eval_ckpt tb_dict = eval_utils .eval_one_epoch( 문자 "/home/a/deepnet/OpenPCDet/tools/eval_utils/eval_utils.py", 제 125행, eval_one_epoch에서 result_str, result_dict = 데이터세트.평가( 文件 "/home/a/deepnet/OpenPCDet/tools/. ./pcdet/datasets/custom/custom_dataset.py”, 제 136절 ap_result_str, ap_dict = kitti_eval(eval_det_annos, eval_gt_annos, self.map_class_to_kitti) 문서 “/home/a/deepnet/OpenPCDet/tools/../pcdet/datasets/custom/custom_dataset.py", 제 127행, 여기 kitti_eval ap_result_str, ap_dict = kitti_eval.get_official_eval_result( 문서 "/home/ a/deepnet/OpenPCDet/tools/../pcdet/datasets/kitti/kitti_object_eval_python/eval.py" ,제 662행, get_official_eval_result current_classes_int.append(name_to_class[curcls]) KeyError: '버스
> 
> 안녕하세요, 저도 같은 문제가 발생했습니다. 이제 해결하셨나요?

In the openpcdet/pcdet/dataset/kitti/kitti_object_eval_python/eval.py part, you can change the class to the class you want to detect,

closed 2024-03-27 by Parkeunseok