open-mmlab/OpenPCDet #1565: AttributeError: 'EasyDict' object has no attribute 'BACKUP_DB_INFO'

Zixiu99, 2024-02-24:
I can confirm that I have generated a valid bin file in gt_database and the DB_INFO_PATH is custom_dbinfos_train.pkl. is there anyone else who is facing the same problem as me?(And also commented out (#filter_by_difficulty: [-1])

Traceback (most recent call last):
  File "tools/train.py", line 230, in <module>
    main()
  File "tools/train.py", line 117, in main
    train_set, train_loader, train_sampler = build_dataloader(
  File "/home/zixiu/Desktop/OpenPCDet/pcdet/datasets/__init__.py", line 57, in build_dataloader
    dataset = __all__[dataset_cfg.DATASET](
  File "/home/zixiu/Desktop/OpenPCDet/pcdet/datasets/custom/custom_dataset.py", line 23, in __init__
    super().__init__(
  File "/home/zixiu/Desktop/OpenPCDet/pcdet/datasets/dataset.py", line 31, in __init__
    self.data_augmentor = DataAugmentor(
  File "/home/zixiu/Desktop/OpenPCDet/pcdet/datasets/augmentor/data_augmentor.py", line 24, in __init__
    cur_augmentor = getattr(self, cur_cfg.NAME)(config=cur_cfg)
  File "/home/zixiu/Desktop/OpenPCDet/pcdet/datasets/augmentor/data_augmentor.py", line 40, in gt_sampling
    db_sampler = database_sampler.DataBaseSampler(
  File "/home/zixiu/Desktop/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py", line 35, in __init__
    sampler_cfg.DB_INFO_PATH[0] = sampler_cfg.BACKUP_DB_INFO['DB_INFO_PATH']
AttributeError: 'EasyDict' object has no attribute 'BACKUP_DB_INFO'

yrbqy2013, 2024-02-27:
Please check if the value for DATA_AUGMENTOR: AUG_CONFIG_LIST: DB_INFO_PATH: in your .yaml file is set to custom_dbinfos_train.pkl.

Zixiu99, 2024-02-28:
> Please check if the value for DATA_AUGMENTOR: AUG_CONFIG_LIST: DB_INFO_PATH: in your .yaml file is set to custom_dbinfos_train.pkl.
The code in my custom_dataset.yaml looks like this, is this correct?
DB_INFO_PATH:         
              - custom_dbinfos_train.pkl

closed 2024-03-07 by Zixiu99

zhu2900000, 2024-12-05:
> Can't pinpoint the cause of the error report at the moment, but provide a sheet of ways to skip the issue, cd to the tools folder and then train. cd tools python train.py --cfg_file cfgs/custom_models/pointrcnn.yaml --batch_size=2 --epochs=30

good answer 
also no problem in guidance
[image]

pangzi-debug, 2025-04-14:
I have encountered this problem also, and it is because I have made a typo in my cfg file which cause a wrong db_info_path resolve result.

pzy01, 2025-05-15:
You can try modifying the DATA_PATH in model.yaml