open-mmlab/OpenPCDet #1573: RuntimeError: Error compiling objects for extension

FrozenWolf-Cyber, 2024-03-03:
I am getting the following error when trying to follow :

```
FAILED: /local/scratch/v_gokul_adethya_t/OpenPCDet/build/temp.linux-x86_64-cpython-38/pcdet/ops/pointnet2/pointnet2_stack/src/voxel_query_gpu.o 
/usr/bin/nvcc  -I/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/lib/python3.8/site-packages/torch/include -I/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/lib/python3.8/site-packages/torch/include/TH -I/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/lib/python3.8/site-packages/torch/include/THC -I/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/include/python3.8 -c -c /local/scratch/v_gokul_adethya_t/OpenPCDet/pcdet/ops/pointnet2/pointnet2_stack/src/voxel_query_gpu.cu -o /local/scratch/v_gokul_adethya_t/OpenPCDet/build/temp.linux-x86_64-cpython-38/pcdet/ops/pointnet2/pointnet2_stack/src/voxel_query_gpu.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1011"' -DTORCH_EXTENSION_NAME=pointnet2_stack_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -gencode=arch=compute_86,code=compute_86 -gencode=arch=compute_86,code=sm_86 -std=c++14
/usr/include/c++/11/bits/std_function.h:435:145: error: parameter packs not expanded with ‘...’:
  435 |         function(_Functor&& __f)
      |                                                                                                                                                 ^ 
/usr/include/c++/11/bits/std_function.h:435:145: note:         ‘_ArgTypes’
/usr/include/c++/11/bits/std_function.h:530:146: error: parameter packs not expanded with ‘...’:
  530 |         operator=(_Functor&& __f)
      |                                                                                                                                                  ^ 
/usr/include/c++/11/bits/std_function.h:530:146: note:         ‘_ArgTypes’
ninja: build stopped: subcommand failed.
Traceback (most recent call last):
  File "/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 1900, in _run_ninja_build
    subprocess.run(
  File "/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/lib/python3.8/subprocess.py", line 516, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['ninja', '-v']' returned non-zero exit status 1.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "setup.py", line 34, in <module>
    setup(
  File "/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/lib/python3.8/site-packages/setuptools/__init__.py", line 103, in setup
    return distutils.core.setup(**attrs)
  File "/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/lib/python3.8/site-packages/setuptools/_distutils/core.py", line 185, in setup
    return run_commands(dist)
  File "/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/lib/python3.8/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
    dist.run_commands()
  File "/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
    self.run_command(cmd)
  File "/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/lib/python3.8/site-packages/setuptools/dist.py", line 989, in run_command
    super().run_command(command)
  File "/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/lib/python3.8/site-packages/setuptools/command/develop.py", line 34, in run
    self.install_for_development()
  File "/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/lib/python3.8/site-packages/setuptools/command/develop.py", line 109, in install_for_development
    self.run_command('build_ext')
  File "/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/lib/python3.8/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
    self.distribution.run_command(command)
  File "/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/lib/python3.8/site-packages/setuptools/dist.py", line 989, in run_command
    super().run_command(command)
  File "/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/lib/python3.8/site-packages/setuptools/command/build_ext.py", line 88, in run
    _build_ext.run(self)
  File "/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/lib/python3.8/site-packages/setuptools/_distutils/command/build_ext.py", line 345, in run
    self.build_extensions()
  File "/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 843, in build_extensions
    build_ext.build_extensions(self)
  File "/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/lib/python3.8/site-packages/setuptools/_distutils/command/build_ext.py", line 467, in build_extensions
    self._build_extensions_serial()
  File "/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/lib/python3.8/site-packages/setuptools/_distutils/command/build_ext.py", line 493, in _build_extensions_serial
    self.build_extension(ext)
  File "/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/lib/python3.8/site-packages/setuptools/command/build_ext.py", line 249, in build_extension
    _build_ext.build_extension(self, ext)
  File "/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/lib/python3.8/site-packages/setuptools/_distutils/command/build_ext.py", line 548, in build_extension
    objects = self.compiler.compile(
  File "/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 658, in unix_wrap_ninja_compile
    _write_ninja_file_and_compile_objects(
  File "/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 1573, in _write_ninja_file_and_compile_objects
    _run_ninja_build(
  File "/shared/home/v_gokul_adethya_t/miniconda3/envs/voxel/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 1916, in _run_ninja_build
    raise RuntimeError(message) from e
RuntimeError: Error compiling objects for extension
(voxel)
```

This is my setup in conda:

```
conda install pytorch==1.13.1 torchvision==0.14.1 torchaudio==0.13.1 pytorch-cuda=11.7 -c pytorch -c nvidia -y
pip install spconv-cu117
python setup.py develop
```

Is there any compatibility issue in spconv version? Does it need to be built from source?

FrozenWolf-Cyber, 2024-03-04:
My gpu config:

```
NVIDIA-SMI 535.154.05
Driver Version: 535.154.05 
CUDA Version: 12.2 
9 x NVIDIA GeForce RTX 3090
```

I am using conda environment, with `python 3.8`

closed 2024-04-19 by github-actions[bot]

TristanCot, 2024-04-24:
I have the same problem, did you manage to solve it?

FrozenWolf-Cyber, 2024-04-25:
Nope