open-mmlab/OpenPCDet #1574: When preparing NuScenes data set: Error ModuleNotFoundError: No module named 'av2'

Tony-syr, 2024-03-04:
av2 is required for Argoverse2 Dataset, however when I try to build nuscenes dataset, it still require the library. I found the because the initialization process requires building the file 

> OpenPCDet/pcdet/datasets/__init__.py

, which import all required datasets. meanwhile, the requirements file doesn't include av2.

reynerlius, 2024-03-12:
have u solved it?
i have the same question when i process the kitti dataset

Tony-syr, 2024-03-13:
@reynerlius Yes, In fact I have not solved it completely, I just commented the line where they import the Argo2Dataset dataset.
So if you don't need to use argo dataset, you can do the same following the same steps:
1- navigate to `OpenPCDet/pcdet/datasets/init.py.`
2- delete or comment the following lines: Line 15 `from .argo2.argo2_dataset import Argo2Dataset` and Line 27 `'Argo2Dataset': Argo2Dataset`

reynerlius, 2024-03-13:
It worked!
Actually I have been puzzled by it for days,Thanks a lot!!

closed 2024-04-27 by github-actions[bot]