open-mmlab/OpenPCDet #1575: Testing voxelnext (voxel_size 0.075) on NuScenes data set gives error "AssertionError: currently only spconv 3D is supported".

Tony-syr, 2024-03-05:
I'm trying to test the VoxelNeXt model provided by this repository with the NuScenes dataset, However, while loading the pre-trained model, it gives the following assertions error : 
File "test.py", line 207, in <module>
    main()
  File "test.py", line 203, in main
    eval_single_ckpt(model, test_loader, args, eval_output_dir, logger, epoch_id, dist_test=dist_test)
  File "test.py", line 61, in eval_single_ckpt
    pre_trained_path=args.pretrained_model)
  File "../pcdet/models/detectors/detector3d_template.py", line 379, in load_params_from_file
    state_dict, update_model_state = self._load_state_dict(model_state_disk, strict=False)
  File "../pcdet/models/detectors/detector3d_template.py", line 346, in _load_state_dict
    assert val.shape.__len__() == 5, 'currently only spconv 3D is supported'
AssertionError: currently only spconv 3D is supported.

---------
My spconv vesion is spconv 1.2.1. 
I printed the val.shape, and I found that one layer shape is _val.shape:  torch.Size([128, 3, 3, 128])_.
so Is there any problem with the pre-trained model itself or something else is wrong?

closed 2024-04-19 by github-actions[bot]

nicoduchene, 2024-04-25:
I've had the same issue recently, any ideas on this ?

cheylong, 2024-09-30:
uncomment this if branch