open-mmlab/OpenPCDet #1579: ModuleNotFoundError: No module named 'pcdet.models.backbones_3d.focal_sparse_conv'

Zixiu99, 2024-03-07:
python tools/train.py --cfg_file tools/cfgs/custom_models/pointrcnn.yaml --batch_size=2 --epochs=30
Traceback (most recent call last):
  File "tools/train.py", line 7, in <module>
    from test import repeat_eval_ckpt
  File "/home/zixiu/Desktop/OpenPCDet/tools/test.py", line 14, in <module>
    from eval_utils import eval_utils
  File "/home/zixiu/Desktop/OpenPCDet/tools/eval_utils/eval_utils.py", line 8, in <module>
    from pcdet.models import load_data_to_gpu
  File "/home/zixiu/anaconda3/envs/openpcdet/lib/python3.8/site-packages/pcdet-0.6.0+255db8f-py3.8-linux-x86_64.egg/pcdet/models/__init__.py", line 6, in <module>
    from .detectors import build_detector
  File "/home/zixiu/anaconda3/envs/openpcdet/lib/python3.8/site-packages/pcdet-0.6.0+255db8f-py3.8-linux-x86_64.egg/pcdet/models/detectors/__init__.py", line 1, in <module>
    from .detector3d_template import Detector3DTemplate
  File "/home/zixiu/anaconda3/envs/openpcdet/lib/python3.8/site-packages/pcdet-0.6.0+255db8f-py3.8-linux-x86_64.egg/pcdet/models/detectors/detector3d_template.py", line 8, in <module>
    from .. import backbones_2d, backbones_3d, dense_heads, roi_heads
  File "/home/zixiu/anaconda3/envs/openpcdet/lib/python3.8/site-packages/pcdet-0.6.0+255db8f-py3.8-linux-x86_64.egg/pcdet/models/backbones_3d/__init__.py", line 4, in <module>
    from .spconv_backbone_focal import VoxelBackBone8xFocal
  File "/home/zixiu/anaconda3/envs/openpcdet/lib/python3.8/site-packages/pcdet-0.6.0+255db8f-py3.8-linux-x86_64.egg/pcdet/models/backbones_3d/spconv_backbone_focal.py", line 7, in <module>
    from .focal_sparse_conv.focal_sparse_conv import FocalSparseConv
ModuleNotFoundError: No module named 'pcdet.models.backbones_3d.focal_sparse_conv'

Zixiu99, 2024-03-07:
Reinstalling pcdet solves the problem.
python setup.py develop

closed 2024-03-07 by Zixiu99

korabelnikov, 2025-08-08:
modern Python/pip fails with       
`
ModuleNotFoundError: No module named 'torch' during installation
`

needed flag to build somehow is `--no-build-isolation`

but even after sucessfull build with 
`
 pip install --no-build-isolation --user .
`
I got original error

Petros626, 2025-11-04:
@korabelnikov solved it?