open-mmlab/OpenPCDet #1580: Much lower detection results on Kitti Pedestrian class + error setting  USE_ROAD_PLANE = True

sinatayebati, 2024-03-09:
I have done an experiment with training SECOND model on kitti, but for some reason when I use the default config at [kitti_dataset.yaml](https://github.com/open-mmlab/OpenPCDet/blob/master/tools/cfgs/dataset_configs/kitti_dataset.yaml), I get the following error:

  File "/home/sina/miniconda3/envs/rd-mae/lib/python3.7/site-packages/torch/utils/data/_utils/fetch.py", line 49, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "../pcdet/datasets/kitti/kitti_dataset.py", line 425, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)
  File "../pcdet/datasets/dataset.py", line 188, in prepare_data
    'gt_boxes_mask': gt_boxes_mask
  File "../pcdet/datasets/augmentor/data_augmentor.py", line 302, in forward
    data_dict = cur_augmentor(data_dict=data_dict)
  File "../pcdet/datasets/augmentor/database_sampler.py", line 498, in __call__
    data_dict, sampled_gt_boxes, total_valid_sampled_dict, sampled_mv_height, sampled_gt_boxes2d
  File "../pcdet/datasets/augmentor/database_sampler.py", line 372, in add_sampled_boxes_to_scene
    sampled_gt_boxes, data_dict['road_plane'], data_dict['calib']
    
This is despite the fact that I'm using the "planes" file for producing data infos as stated in the GETTING STARTED. After facing this error, I set the [USE_ROAD_PLANE](https://github.com/open-mmlab/OpenPCDet/blob/255db8f02a8bd07211d2c91f54602d63c4c93356/tools/cfgs/dataset_configs/kitti_dataset.yaml#L23) to "False" to be able to execute the training. Although the training goes through, the AP results are very different from the report in particular for "pedestrian" class.

1 - does setting USE_ROAD_PLANE to False in DATA AUGMENTATION cause this much of change in detection accuracy of pedestrian class?
2 - what could be the cause of facing the error in in the first place when using the default config and can I resolve this error?

Following is the log of my training.

2024-03-07 08:39:22,448   INFO  **********************Start logging**********************
2024-03-07 08:39:22,448   INFO  CUDA_VISIBLE_DEVICES=ALL
2024-03-07 08:39:22,448   INFO  Training in distributed mode : total_batch_size: 8
2024-03-07 08:39:22,448   INFO  cfg_file         cfgs/kitti_models/second.yaml
2024-03-07 08:39:22,448   INFO  batch_size       4
2024-03-07 08:39:22,448   INFO  epochs           80
2024-03-07 08:39:22,448   INFO  workers          4
2024-03-07 08:39:22,448   INFO  extra_tag        default
2024-03-07 08:39:22,448   INFO  ckpt             None
2024-03-07 08:39:22,448   INFO  pretrained_model None
2024-03-07 08:39:22,448   INFO  launcher         pytorch
2024-03-07 08:39:22,448   INFO  tcp_port         18888
2024-03-07 08:39:22,448   INFO  sync_bn          False
2024-03-07 08:39:22,448   INFO  fix_random_seed  False
2024-03-07 08:39:22,448   INFO  ckpt_save_interval 1
2024-03-07 08:39:22,448   INFO  local_rank       0
2024-03-07 08:39:22,448   INFO  max_ckpt_save_num 15
2024-03-07 08:39:22,448   INFO  merge_all_iters_to_one_epoch False
2024-03-07 08:39:22,448   INFO  set_cfgs         None
2024-03-07 08:39:22,448   INFO  max_waiting_mins 0
2024-03-07 08:39:22,448   INFO  start_epoch      0
2024-03-07 08:39:22,448   INFO  num_epochs_to_eval 10
2024-03-07 08:39:22,448   INFO  save_to_file     False
2024-03-07 08:39:22,448   INFO  use_tqdm_to_record False
2024-03-07 08:39:22,448   INFO  logger_iter_interval 50
2024-03-07 08:39:22,449   INFO  ckpt_save_time_interval 300
2024-03-07 08:39:22,449   INFO  wo_gpu_stat      False
2024-03-07 08:39:22,449   INFO  use_amp          False
2024-03-07 08:39:22,449   INFO  cfg.ROOT_DIR: /hdd_10tb/sina/Radial_MAE
2024-03-07 08:39:22,449   INFO  cfg.LOCAL_RANK: 0
2024-03-07 08:39:22,449   INFO  cfg.CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist']
2024-03-07 08:39:22,449   INFO  ----------- DATA_CONFIG -----------
2024-03-07 08:39:22,449   INFO  cfg.DATA_CONFIG.DATASET: KittiDataset
2024-03-07 08:39:22,449   INFO  cfg.DATA_CONFIG.DATA_PATH: ../data/kitti
2024-03-07 08:39:22,449   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [0, -40, -3, 70.4, 40, 1]
2024-03-07 08:39:22,449   INFO  ----------- DATA_SPLIT -----------
2024-03-07 08:39:22,449   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: train
2024-03-07 08:39:22,449   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: val
2024-03-07 08:39:22,449   INFO  ----------- INFO_PATH -----------
2024-03-07 08:39:22,449   INFO  cfg.DATA_CONFIG.INFO_PATH.train: ['kitti_infos_train.pkl']
2024-03-07 08:39:22,449   INFO  cfg.DATA_CONFIG.INFO_PATH.test: ['kitti_infos_val.pkl']
2024-03-07 08:39:22,449   INFO  cfg.DATA_CONFIG.GET_ITEM_LIST: ['points']
2024-03-07 08:39:22,449   INFO  cfg.DATA_CONFIG.FOV_POINTS_ONLY: True
2024-03-07 08:39:22,449   INFO  ----------- DATA_AUGMENTOR -----------
2024-03-07 08:39:22,449   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder']
2024-03-07 08:39:22,449   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'gt_sampling', 'USE_ROAD_PLANE': False, 'DB_INFO_PATH': ['kitti_dbinfos_train.pkl'], 'PREPARE': {'filter_by_min_points': ['Car:5', 'Pedestrian:5', 'Cyclist:5'], 'filter_by_difficulty': [-1]}, 'SAMPLE_GROUPS': ['Car:20', 'Pedestrian:15', 'Cyclist:15'], 'NUM_POINT_FEATURES': 4, 'DATABASE_WITH_FAKELIDAR': False, 'REMOVE_EXTRA_WIDTH': [0.0, 0.0, 0.0], 'LIMIT_WHOLE_SCENE': True}, {'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.78539816, 0.78539816]}, {'NAME': 'random_world_scaling', 'WORLD_SCALE_RANGE': [0.95, 1.05]}]
2024-03-07 08:39:22,449   INFO  ----------- POINT_FEATURE_ENCODING -----------
2024-03-07 08:39:22,449   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
[... 1054 lines cut ...]
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:86.8128, 78.7895, 75.1452
bev  AP:82.7761, 71.7918, 67.8346
3d   AP:82.0577, 69.2279, 64.5570
aos  AP:86.53, 78.01, 74.36
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:91.2360, 79.9652, 75.9099
bev  AP:85.2570, 72.5337, 67.5499
3d   AP:84.4018, 70.0027, 65.1040
aos  AP:90.86, 79.08, 75.05
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:86.8128, 78.7895, 75.1452
bev  AP:91.0714, 76.6888, 72.8220
3d   AP:91.0714, 76.6888, 72.8220
aos  AP:86.53, 78.01, 74.36
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:91.2360, 79.9652, 75.9099
bev  AP:91.6310, 77.6228, 73.5715
3d   AP:91.6310, 77.6228, 73.5715
aos  AP:90.86, 79.08, 75.05

2024-03-07 13:45:14,843   INFO  Result is saved to /hdd_10tb/sina/Radial_MAE/output/kitti_models/second/default/eval/eval_with_train/epoch_80/val
2024-03-07 13:45:14,844   INFO  ****************Evaluation done.*****************
2024-03-07 13:45:14,864   INFO  Epoch 80 has been evaluated
2024-03-07 13:45:44,895   INFO  **********************End evaluation kitti_models/second(default)**********************

sinatayebati, 2024-03-12:
It's also worth noting that when I evaluate using ckpt published by OpenPCDet, I get the exact same accuracy as reported. Thus, I assume it's a problem of data infos. This problem only happens when I train from scratch. 

Another observation is that when I train any of models from scratch on other datasets including Waymo and NuScene, I can reproduce the same exact accuracy. Thus, this observation also tells me it's safe to assume the environment is not an issue as well. 

I'm still struggling with this issue and cannot find the root cause why I cannot reproduce your reported results by training from scratch on KITTI, so any hints or suggestions would be greatly appreciated.

Regarding my environment, I'm using pytorch=1.10.2, spconv=2.3.6, and the rest is attached bellow:

name: rd-mae
channels:
  - pytorch3d
  - plotly
  - bottler
  - iopath
  - fvcore
  - pytorch
  - conda-forge
  - defaults
dependencies:
  - cudatoolkit-dev=11.3.1
  - plotly=5.8.1
  - plotly_express=0.4.1
  - python=3.7.13
  - pytorch=1.10.2
  - pytorch3d=0.6.2
  - pyyaml=6.0
  - setuptools=62.3.3
  - torchvision=0.11.3
  - tqdm=4.64.0
  - pip
  - ipykernel
  - ipython
  - pip:
    - argparse==1.4.0
    - gdown==4.4.0
    - h5py==3.7.0
    - easydict
    - matplotlib==3.5.2
    - matplotlib-inline==0.1.3
    - open3d==0.15.2
    - opencv-python==4.6.0.66
    - pandas==1.3.5
    - tensorboardx==2.5.1
    - timm==0.4.5
    - tinycss2==1.1.1
    - tomlkit==0.11.0
    - traitlets==5.2.2.post1
    - transforms3d==0.3.1
    - ninja 
    - faiss-cpu
    - wandb

closed 2024-04-26 by github-actions[bot]

zhangye1111, 2024-06-19:
> 还值得注意的是，当我使用 OpenPCDet 发布的 ckpt 进行评估时，我得到的准确率与报告的完全相同。因此，我认为这是数据信息的问题。这个问题只发生在我从头开始训练的时候。
> 
> 另一个观察结果是，当我在其他数据集（包括 Waymo 和 NuScene）上从头训练任何模型时，我都可以重现完全相同的准确率。因此，这个观察结果也告诉我，可以安全地假设环境也不是问题。
> 
> 我仍在努力解决这个问题，但找不到导致我无法通过在 KITTI 上从头开始训练来重现您报告的结果的根本原因，因此，任何提示或建议都将不胜感激。
> 
> 关于我的环境，我使用的是 pytorch=1.10.2、spconv=2.3.6，其余部分附在下面：
> 
> 名称：rd-mae 频道：
> 
> * pytorch3d
> * 阴谋地
> * 瓶装商
> * 免疫学
> * 核心数
> * pythorch
> * 康达锻造公司
> * 默认
>   依赖项：
> * cudatoolkit-dev=11.3.1
> * 绘图=5.8.1
> * plotly_express=0.4.1
> * 蟒蛇=3.7.13
> * 火炬=1.10.2
> * pytorch3d=0.6.2
> * pyyaml=6.0
> * 设置工具=62.3.3
> * torchvision=0.11.3
> * tqdm=4.64.0
> * 点子
> * 内核
> * Python
> * 点子：
>   
>   * argparse==1.4.0
>   * gdown==4.4.0
>   * h5py==3.7.0
>   * 易词典
>   * matplotlib==3.5.2
>   * matplotlib-inline==0.1.3
>   * open3d==0.15.2
>   * opencv-python==4.6.0.66
>   * 熊猫==1.3.5
>   * tensorboardx==2.5.1
>   * timm==0.4.5
>   * tinycss2==1.1.1
>   * tomlkit==0.11.0
>   * traitlets==5.2.2.post1
>   * transforms3d==0.3.1
>   * 忍者
>   * faiss-cpu
>   * 万德布

May I ask if you have solved this problem? I get low pedestrian detection as well.

YayoYY, 2026-04-14:
> May I ask if you have solved this problem? I get low pedestrian detection as well.

I have the same problem too, have you solved it?