open-mmlab/OpenPCDet #1583: Running the Demo throws RuntimeError

JensStaub, 2024-03-12:
When trying to run the demo.py using the command

`~/Development/OpenPCDet/tools$ python3 demo.py --cfg_file cfgs/kitti_models/pointpillar.yaml --ckpt /home/data/Development/models/pointpillar_7728.pth --data_path /home/data/Development/datasets/kitti/velodyne_points/001005.bin`

I get the following Error:

`
Traceback (most recent call last):
  File "/home/user/Development/OpenPCDet/tools/demo.py", line 18, in <module>
    from pcdet.datasets import DatasetTemplate
  File "/home/user/Development/OpenPCDet/pcdet/datasets/__init__.py", line 15, in <module>
    from .argo2.argo2_dataset import Argo2Dataset
  File "/home/user/Development/OpenPCDet/pcdet/datasets/argo2/argo2_dataset.py", line 15, in <module>
    from .argo2_utils.so3 import yaw_to_quat, quat_to_yaw
  File "/home/user/Development/OpenPCDet/pcdet/datasets/argo2/argo2_utils/so3.py", line 10, in <module>
    def quat_to_mat(quat_wxyz: Tensor) -> Tensor:
  File "/usr/local/lib/python3.10/dist-packages/torch/jit/_script.py", line 1395, in script
    fn = torch._C._jit_script_compile(
  File "/usr/local/lib/python3.10/dist-packages/torch/jit/_recursive.py", line 1003, in try_compile_fn
    return torch.jit.script(fn, _rcb=rcb)
  File "/usr/local/lib/python3.10/dist-packages/torch/jit/_script.py", line 1395, in script
    fn = torch._C._jit_script_compile(
RuntimeError: 
cannot statically infer the expected size of a list in this context:
  File "/usr/local/lib/python3.10/dist-packages/kornia/geometry/conversions.py", line 553

    # this slightly awkward construction of the output shape is to satisfy torchscript
    output_shape = [*list(quaternion.shape[:-1]), 3, 3]
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~ <--- HERE
    matrix = matrix_flat.reshape(output_shape)
'quaternion_to_rotation_matrix' is being compiled since it was called from 'quat_to_mat'
  File "/home/user/Development/OpenPCDet/pcdet/datasets/argo2/argo2_utils/so3.py", line 19
        (...,3,3) 3D rotation matrices.
    """
    return C.quaternion_to_rotation_matrix(
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        quat_wxyz, order=C.QuaternionCoeffOrder.WXYZ
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ <--- HERE
    )
`

I am testing on the following system:
- Ubuntu 22.04
- python3.10
- CUDA 12.3
- torch 2.2.1
- kornia 0.7.1

I downloaded the KITTI dataset and the checkpoints from the website linked in the README.

Additionally I have also tried it with the Docker image built from the `cu116_Dockerfile`, but with the same error.

Does anybody know why this error is popping up?

JensStaub, 2024-03-14:
As mentioned here:
https://github.com/open-mmlab/OpenPCDet/pull/1522
I downgraded "kornia" to version 0.5.8 and it is working now.

closed 2024-03-14 by JensStaub

llz666ha, 2025-03-20:
> When trying to run the demo.py using the command尝试使用命令运行 demo.py 时
> 
> `~/Development/OpenPCDet/tools$ python3 demo.py --cfg_file cfgs/kitti_models/pointpillar.yaml --ckpt /home/data/Development/models/pointpillar_7728.pth --data_path /home/data/Development/datasets/kitti/velodyne_points/001005.bin`
> 
> I get the following Error:我收到以下错误：
> 
> ` Traceback (most recent call last):回溯 （最近调用最后）： File "/home/user/Development/OpenPCDet/tools/demo.py", line 18, in 文件 “/home/user/Development/OpenPCDet/tools/demo.py”，第 18 行，在 from pcdet.datasets import DatasetTemplatefrom pcdet.datasets 导入 DatasetTemplate File "/home/user/Development/OpenPCDet/pcdet/datasets/**init**.py", line 15, in 文件 “/home/user/Development/OpenPCDet/pcdet/datasets/**init**.py”，第 15 行，在 from .argo2.argo2_dataset import Argo2Dataset File "/home/user/Development/OpenPCDet/pcdet/datasets/argo2/argo2_dataset.py", line 15, in 文件 “/home/user/Development/OpenPCDet/pcdet/datasets/argo2/argo2_dataset.py”，第 15 行，在 from .argo2_utils.so3 import yaw_to_quat, quat_to_yawfrom .argo2_utils.so3 import yaw_to_quat， quat_to_yaw File "/home/user/Development/OpenPCDet/pcdet/datasets/argo2/argo2_utils/so3.py", line 10, in 文件 “/home/user/Development/OpenPCDet/pcdet/datasets/argo2/argo2_utils/so3.py”，第 10 行，在 def quat_to_mat(quat_wxyz: Tensor) -> Tensor:def quat_to_mat（quat_wxyz： 张紧器） -> 张紧器： File "/usr/local/lib/python3.10/dist-packages/torch/jit/_script.py", line 1395, in script脚本中的文件 “/usr/local/lib/python3.10/dist-packages/torch/jit/_script.py”，第 1395 行 fn = torch._C._jit_script_compile(fn = torch._C._jit_script_compile（ File "/usr/local/lib/python3.10/dist-packages/torch/jit/_recursive.py", line 1003, in try_compile_fn文件 “/usr/local/lib/python3.10/dist-packages/torch/jit/_recursive.py”，第 1003 行，try_compile_fn return torch.jit.script(fn, _rcb=rcb)返回 torch.jit.script（fn， _rcb=rcb） File "/usr/local/lib/python3.10/dist-packages/torch/jit/_script.py", line 1395, in script脚本中的文件 “/usr/local/lib/python3.10/dist-packages/torch/jit/_script.py”，第 1395 行 fn = torch._C._jit_script_compile(fn = torch._C._jit_script_compile（ RuntimeError:  RuntimeError 错误： cannot statically infer the expected size of a list in this context:无法在此上下文中静态推断列表的预期大小： File "/usr/local/lib/python3.10/dist-packages/kornia/geometry/conversions.py", line 553文件 “/usr/local/lib/python3.10/dist-packages/kornia/geometry/conversions.py”，第 553 行
> 
> ```
> # this slightly awkward construction of the output shape is to satisfy torchscript
> output_shape = [*list(quaternion.shape[:-1]), 3, 3]
>                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~ <--- HERE
> matrix = matrix_flat.reshape(output_shape)
> ```
> 
> 'quaternion_to_rotation_matrix' is being compiled since it was called from 'quat_to_mat'“quaternion_to_rotation_matrix”是从“quat_to_mat”调用的，因此正在编译 File "/home/user/Development/OpenPCDet/pcdet/datasets/argo2/argo2_utils/so3.py", line 19文件 “/home/user/Development/OpenPCDet/pcdet/datasets/argo2/argo2_utils/so3.py”，第 19 行 (...,3,3) 3D rotation matrices.（...，3,3） 3D 旋转矩阵。 """ return C.quaternion_to_rotation_matrix(返回 C.quaternion_to_rotation_matrix（ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ quat_wxyz, order=C.QuaternionCoeffOrder.WXYZquat_wxyz，order=C.QuaternionCoeffOrder.WXYZ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ <--- HERE~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ <--- 这里 ) `
> 
> I am testing on the following system:我正在以下系统上进行测试：
> 
> * Ubuntu 22.04  Ubuntu 22.04 版本
> * python3.10  Python3.10 的
> * CUDA 12.3  12.3 奇迹
> * torch 2.2.1  火炬 2.2.1
> * kornia 0.7.1  檐口 0.7.1
> 
> I downloaded the KITTI dataset and the checkpoints from the website linked in the README.我从 README 中链接的网站下载了 KITTI 数据集和检查点。
> 
> Additionally I have also tried it with the Docker image built from the `cu116_Dockerfile`, but with the same error.此外，我还尝试了从 `cu116_Dockerfile` 构建的 Docker 镜像，但出现相同的错误。
> 
> Does anybody know why this error is popping up?有人知道为什么会弹出这个错误吗？
e

> When trying to run the demo.py using the command尝试使用命令运行 demo.py 时
> 
> `~/Development/OpenPCDet/tools$ python3 demo.py --cfg_file cfgs/kitti_models/pointpillar.yaml --ckpt /home/data/Development/models/pointpillar_7728.pth --data_path /home/data/Development/datasets/kitti/velodyne_points/001005.bin`
> 
> I get the following Error:我收到以下错误：
> 
> ` Traceback (most recent call last):回溯 （最近调用最后）： File "/home/user/Development/OpenPCDet/tools/demo.py", line 18, in 文件 “/home/user/Development/OpenPCDet/tools/demo.py”，第 18 行，在 from pcdet.datasets import DatasetTemplatefrom pcdet.datasets 导入 DatasetTemplate File "/home/user/Development/OpenPCDet/pcdet/datasets/**init**.py", line 15, in 文件 “/home/user/Development/OpenPCDet/pcdet/datasets/**init**.py”，第 15 行，在 from .argo2.argo2_dataset import Argo2Dataset File "/home/user/Development/OpenPCDet/pcdet/datasets/argo2/argo2_dataset.py", line 15, in 文件 “/home/user/Development/OpenPCDet/pcdet/datasets/argo2/argo2_dataset.py”，第 15 行，在 from .argo2_utils.so3 import yaw_to_quat, quat_to_yawfrom .argo2_utils.so3 import yaw_to_quat， quat_to_yaw File "/home/user/Development/OpenPCDet/pcdet/datasets/argo2/argo2_utils/so3.py", line 10, in 文件 “/home/user/Development/OpenPCDet/pcdet/datasets/argo2/argo2_utils/so3.py”，第 10 行，在 def quat_to_mat(quat_wxyz: Tensor) -> Tensor:def quat_to_mat（quat_wxyz： 张紧器） -> 张紧器： File "/usr/local/lib/python3.10/dist-packages/torch/jit/_script.py", line 1395, in script脚本中的文件 “/usr/local/lib/python3.10/dist-packages/torch/jit/_script.py”，第 1395 行 fn = torch._C._jit_script_compile(fn = torch._C._jit_script_compile（ File "/usr/l [... cut ...]
> * CUDA 12.3  12.3 奇迹
> * torch 2.2.1  火炬 2.2.1
> * kornia 0.7.1  檐口 0.7.1
> 
> I downloaded the KITTI dataset and the checkpoints from the website linked in the README.我从 README 中链接的网站下载了 KITTI 数据集和检查点。
> 
> Additionally I have also tried it with the Docker image built from the `cu116_Dockerfile`, but with the same error.此外，我还尝试了从 `cu116_Dockerfile` 构建的 Docker 镜像，但出现相同的错误。
> 
> Does anybody know why this error is popping up?有人知道为什么会弹出这个错误吗？

excuse me,How to build this project on Ubuntu 22.04?please