open-mmlab/OpenPCDet #1586: Multi GPU training Error

ammaryasirnaich, 2024-03-13:
i am using the dist_train.sh to train pointpiller model on 2 GPUS. But i am getting an error as below

```
+ NGPUS=2
+ PY_ARGS='--cfg_file /workspace/OpenPCDet/tools/cfgs/waymo_models/pointpillar_1x.yaml --batch_size 2 --epochs 40'
+ true
+ PORT=51563
++ nc -z 127.0.0.1 51563
++ echo 1
+ status=1
+ '[' 1 '!=' 0 ']'
+ break
+ echo 51563
51563
+ python -m torch.distributed.launch --nproc_per_node=2 --rdzv_endpoint=localhost:51563 train.py --launcher pytorch --cfg_file /workspace/OpenPCDet/tools/cfgs/waymo_models/pointpillar_1x.yaml --batch_size 2 --epochs 40
/pcdet/lib/python3.8/site-packages/torch/distributed/launch.py:183: FutureWarning: The module torch.distributed.launch is deprecated
and will be removed in future. Use torchrun.
Note that --use-env is set by default in torchrun.
If your script expects `--local-rank` argument to be set, please
change it to read from `os.environ['LOCAL_RANK']` instead. See 
https://pytorch.org/docs/stable/distributed.html#launch-utility for 
further instructions

  warnings.warn(
[2024-03-13 14:08:09,269] torch.distributed.run: [WARNING] 
[2024-03-13 14:08:09,269] torch.distributed.run: [WARNING] *****************************************
[2024-03-13 14:08:09,269] torch.distributed.run: [WARNING] Setting OMP_NUM_THREADS environment variable for each process to be 1 in default, to avoid your system being overloaded, please further tune the variable for optimal performance in your application as needed. 
[2024-03-13 14:08:09,269] torch.distributed.run: [WARNING] *****************************************
usage: train.py [-h] [--cfg_file CFG_FILE] [--batch_size BATCH_SIZE] [--epochs EPOCHS] [--workers WORKERS] [--extra_tag EXTRA_TAG]
                [--ckpt CKPT] [--pretrained_model PRETRAINED_MODEL] [--launcher {none,pytorch,slurm}] [--tcp_port TCP_PORT] [--sync_bn]
                [--fix_random_seed] [--ckpt_save_interval CKPT_SAVE_INTERVAL] [--local_rank LOCAL_RANK]
                [--max_ckpt_save_num MAX_CKPT_SAVE_NUM] [--merge_all_iters_to_one_epoch] [--set ...]
                [--max_waiting_mins MAX_WAITING_MINS] [--start_epoch START_EPOCH] [--num_epochs_to_eval NUM_EPOCHS_TO_EVAL]
                [--save_to_file] [--use_tqdm_to_record] [--logger_iter_interval LOGGER_ITER_INTERVAL]
                [--ckpt_save_time_interval CKPT_SAVE_TIME_INTERVAL] [--wo_gpu_stat] [--use_amp]
usage: train.py [-h] [--cfg_file CFG_FILE] [--batch_size BATCH_SIZE] [--epochs EPOCHS] [--workers WORKERS] [--extra_tag EXTRA_TAG]
                [--ckpt CKPT] [--pretrained_model PRETRAINED_MODEL] [--launcher {none,pytorch,slurm}] [--tcp_port TCP_PORT] [--sync_bn]
                [--fix_random_seed] [--ckpt_save_interval CKPT_SAVE_INTERVAL] [--local_rank LOCAL_RANK]
                [--max_ckpt_save_num MAX_CKPT_SAVE_NUM] [--merge_all_iters_to_one_epoch] [--set ...]
                [--max_waiting_mins MAX_WAITING_MINS] [--start_epoch START_EPOCH] [--num_epochs_to_eval NUM_EPOCHS_TO_EVAL]
                [--save_to_file] [--use_tqdm_to_record] [--logger_iter_interval LOGGER_ITER_INTERVAL]
                [--ckpt_save_time_interval CKPT_SAVE_TIME_INTERVAL] [--wo_gpu_stat] [--use_amp]
train.py: error: unrecognized arguments: --local-rank=0
train.py: error: unrecognized arguments: --local-rank=1
[2024-03-13 14:08:19,304] torch.distributed.elastic.multiprocessing.api: [ERROR] failed (exitcode: 2) local_rank: 0 (pid: 28415) of binary: /pcdet/bin/python
Traceback (most recent call last):
  File "/import/linux/python/3.8.2/lib/python3.8/runpy.py", line 193, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/import/linux/python/3.8.2/lib/python3.8/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/pcdet/lib/python3.8/site-packages/torch/distributed/launch.py", line 198, in <module>
    main()
  File "/pcdet/lib/python3.8/site-packages/torch/distributed/launch.py", line 194, in main
    launch(args)
  File "/pcdet/lib/python3.8/site-packages/torch/distributed/launch.py", line 179, in launch
    run(args)
  File "/pcdet/lib/python3.8/site-packages/torch/distributed/run.py", line 803, in run
    elastic_launch(
  File "/pcdet/lib/python3.8/site-packages/torch/distributed/launcher/api.py", line 135, in __call__
    return launch_agent(self._config, self._entrypoint, list(args))
  File "/pcdet/lib/python3.8/site-packages/torch/distributed/launcher/api.py", line 268, in launch_agent
    raise ChildFailedError(
torch.distributed.elastic.multiprocessing.errors.ChildFailedError: 
============================================================
train.py FAILED
------------------------------------------------------------
Failures:
[1]:
  time      : 2024-03-13_14:08:19
  host      : durham
  rank      : 1 (local_rank: 1)
  exitcode  : 2 (pid: 28416)
  error_file: <N/A>
  traceback : To enable traceback see: https://pytorch.org/docs/stable/elastic/errors.html
------------------------------------------------------------
Root Cause (first observed failure):
[0]:
  time      : 2024-03-13_14:08:19
  host      : durham
  rank      : 0 (local_rank: 0)
  exitcode  : 2 (pid: 28415)
  error_file: <N/A>
  traceback : To enable traceback see: https://pytorch.org/docs/stable/elastic/errors.html
============================================================
```
I would appreciate it if anyone can point to where i am getting it wrong. Thanks

BiboyQG, 2024-03-26:
Perhaps because of this:
[image]

ammaryasirnaich, 2024-03-26:
> Perhaps because of this: <img alt="image" width="846" src="https://private-user-images.githubusercontent.com/125724218/316750951-f26a8129-18e2-45a0-93ad-f0918c0fe1d2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE0NDg4NzEsIm5iZiI6MTcxMTQ0ODU3MSwicGF0aCI6Ii8xMjU3MjQyMTgvMzE2NzUwOTUxLWYyNmE4MTI5LTE4ZTItNDVhMC05M2FkLWYwOTE4YzBmZTFkMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMyNlQxMDIyNTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wNTZkZjcyNWI5NzljM2YwNzUzNzk0ZGE3Njc4ZDljMmNkMDhiODFmOTZiNDE0MTEwNjc3Yzk0MmExMTkyZjEwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.0wEZlSjAdu9o3tZPUXS-w5HF0Fp1gKk-6nZqYyPjFzM">

Thanks for responding, any idea what should i do to solve it? I literally don't know what to do ahead

BiboyQG, 2024-03-27:
> Thanks for responding, any idea what should i do to solve it? I literally don't know what to do ahead

Yes, I think you can change the argument parser in the tools/train.py from "--local_rank" to "--local_rank" like this:
[image]
After the change, you may be good to go.

ammaryasirnaich, 2024-03-27:
> banghao1023

Sure it will be great, thanks for adding me to wechat, also many thanks for the help really appreciate it

closed 2024-05-11 by github-actions[bot]