open-mmlab/OpenPCDet #1587: Find A bug in transfusion_head.py code, in case that pc_range is rectangle

shuaishi1991, 2024-03-22:
When I use the bevfusion model which integrated into this project, I find that when setting the pc_range to rectangle, then the model training process does not converge well. After examination, I found a bug in the "OpenPCDet/pcdet/models/dense_heads /transfusion_head.py" file. The specific errors is on line 119-121 as follows:

        x_size = self.grid_size[0] // self.feature_map_stride
        y_size = self.grid_size[1] // self.feature_map_stride
        self.bev_pos = self.create_2D_grid(x_size, y_size) 

those codes setting the coordinate order of x and y incorrectly. When the pc_range is a square, this situation will not produce an error, but when it is a rectangle, it will be exposed. the correct code should be as forllows:

        x_size = self.grid_size[0] // self.feature_map_stride
        y_size = self.grid_size[1] // self.feature_map_stride
        self.bev_pos = self.create_2D_grid(y_size, x_size)     # set the y_size to first, and x_size to second

I hope the author pays attention to this issue and conducts verification. Thank you！

closed 2024-05-06 by github-actions[bot]

Youngtrue, 2024-08-14:
This is indeed and exists also in mmdet3d!! Applying the revision solves the training problem in the custom dataset. 
Thanks very much for the report!

xpyqiubai, 2025-04-21:
> Hello@chenshi3
> 
> When I use the bevfusion model which integrated into this project, I find that when setting the pc_range to rectangle, then the model training process does not converge well. After examination, I found a bug in the "OpenPCDet/pcdet/models/dense_heads /transfusion_head.py" file. The specific errors is on line 119-121 as follows:
> 
> ```
>     x_size = self.grid_size[0] // self.feature_map_stride
>     y_size = self.grid_size[1] // self.feature_map_stride
>     self.bev_pos = self.create_2D_grid(x_size, y_size) 
> ```
> 
> those codes setting the coordinate order of x and y incorrectly. When the pc_range is a square, this situation will not produce an error, but when it is a rectangle, it will be exposed. the correct code should be as forllows:
> 
> ```
>     x_size = self.grid_size[0] // self.feature_map_stride
>     y_size = self.grid_size[1] // self.feature_map_stride
>     self.bev_pos = self.create_2D_grid(y_size, x_size)     # set the y_size to first, and x_size to second
> ```
> 
> I hope the author pays attention to this issue and conducts verification. Thank you！

May I ask if the detection indicators in the rectangular area are normal after modifying this position? Are there any other places that need to be modified? Thank you!

xpyqiubai, 2025-04-21:
> This is indeed and exists also in mmdet3d!! Applying the revision solves the training problem in the custom dataset. Thanks very much for the report!

May I ask if the detection indicators in the rectangular area are normal after modifying this position? Are there any other places that need to be modified? Thank you!

shuaishi1991, 2025-05-02:
> May I ask if the detection indicators in the rectangular area are normal after modifying this position? Are there any other places that need to be modified? Thank you!

No further changes are required

wcyjerry, 2025-05-30:
@shuaishi1991 Hi, for rectangle, I think there should be another change in here:def get_targets_single(self, gt_bboxes_3d, gt_labels_3d, preds_dict, batch_idx):,
seems like also misunderstand the h and w