open-mmlab/OpenPCDet #1594: GPU out of memory when doing evaluation

SimonSongg, 2024-03-28:
Hi, I trained CenterPoint on Waymo dataset. Everything looks good. But when I tried to evaluate it, tensorflow occupied all of my GPU memory and I got a Illegal instruction or segmentation fault. I don't think this step needs that much GPU memory, as the inference in PyTorch has finished. Any suggestions are appreciated.

Specs: RTX 4070 Ti Super 16G, CUDA 11.8

`2024-03-28 19:10:56,850   INFO  *************** EPOCH 30 EVALUATION *****************
eval:   0%|                                                               | 0/400 [00:00<?, ?it/s]../pcdet/ops/iou3d_nms/iou3d_nms_utils.py:66: UserWarning: The torch.cuda.*DtypeTensor constructors are no longer recommended. It's best to use methods such as torch.tensor(data, dtype=*, device='cuda') to create tensors. (Triggered internally at ../torch/csrc/tensor/python_tensor.cpp:83.)
  overlaps_bev = torch.cuda.FloatTensor(torch.Size((boxes_a.shape[0], boxes_b.shape[0]))).zero_()  # (N, M)
eval: 100%|██████████████████████| 400/400 [00:15<00:00, 25.50it/s, recall_0.3=(0, 15664) / 18167]
2024-03-28 19:11:12,537   INFO  *************** Performance of EPOCH 30 *****************
2024-03-28 19:11:12,537   INFO  Generate label finished(sec_per_example: 0.0392 second).
2024-03-28 19:11:12,537   INFO  recall_roi_0.3: 0.000000
2024-03-28 19:11:12,537   INFO  recall_rcnn_0.3: 0.862223
2024-03-28 19:11:12,537   INFO  recall_roi_0.5: 0.000000
2024-03-28 19:11:12,537   INFO  recall_rcnn_0.5: 0.792811
2024-03-28 19:11:12,537   INFO  recall_roi_0.7: 0.000000
2024-03-28 19:11:12,537   INFO  recall_rcnn_0.7: 0.508174
2024-03-28 19:11:12,537   INFO  Average predicted number of objects(400 samples): 216.525
2024-03-28 19:11:12.579766: I tensorflow/compiler/xla/stream_executor/cuda/cuda_gpu_executor.cc:982] could not open file to read NUMA node: /sys/bus/pci/devices/0000:01:00.0/numa_node
Your kernel may have been built without NUMA support.
2024-03-28 19:11:12.580574: I tensorflow/compiler/xla/stream_executor/cuda/cuda_gpu_executor.cc:982] could not open file to read NUMA node: /sys/bus/pci/devices/0000:01:00.0/numa_node
Your kernel may have been built without NUMA support.
2024-03-28 19:11:12.580871: I tensorflow/compiler/xla/stream_executor/cuda/cuda_gpu_executor.cc:982] could not open file to read NUMA node: /sys/bus/pci/devices/0000:01:00.0/numa_node
Your kernel may have been built without NUMA support.
2024-03-28 19:11:12.582934: I tensorflow/compiler/xla/stream_executor/cuda/cuda_gpu_executor.cc:982] could not open file to read NUMA node: /sys/bus/pci/devices/0000:01:00.0/numa_node
Your kernel may have been built without NUMA support.
2024-03-28 19:11:12.583995: I tensorflow/compiler/xla/stream_executor/cuda/cuda_gpu_executor.cc:982] could not open file to read NUMA node: /sys/bus/pci/devices/0000:01:00.0/numa_node
Your kernel may have been built without NUMA support.
2024-03-28 19:11:12.584643: I tensorflow/compiler/xla/stream_executor/cuda/cuda_gpu_executor.cc:982] could not open file to read NUMA node: /sys/bus/pci/devices/0000:01:00.0/numa_node
Your kernel may have been built without NUMA support.
2024-03-28 19:11:12.585362: I tensorflow/compiler/xla/stream_executor/cuda/cuda_gpu_executor.cc:982] could not open file to read NUMA node: /sys/bus/pci/devices/0000:01:00.0/numa_node
Your kernel may have been built without NUMA support.
2024-03-28 19:11:12.585795: I tensorflow/compiler/xla/stream_executor/cuda/cuda_gpu_executor.cc:982] could not open file to read NUMA node: /sys/bus/pci/devices/0000:01:00.0/numa_node
Your kernel may have been built without NUMA support.
2024-03-28 19:11:12.585811: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1722] Could not identify NUMA node of platform GPU id 0, defaulting to 0.  Your kernel may not have been built with NUMA support.
2024-03-28 19:11:12.586111: I tensorflow/compiler/xla/stream_executor/cuda/cuda_gpu_executor.cc:982] could not open file to read NUMA node: /sys/bus/pci/devices/0000:01:00.0/numa_node
Your kernel may have been built without NUMA support.
2024-03-28 19:11:12.586145: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1635] Created device /job:localhost/replica:0/task:0/device:GPU:0 with 12853 MB memory:  -> device: 0, name: NVIDIA GeForce RTX 4070 Ti SUPER, pci bus id: 0000:01:00.0, compute capability: 8.9
Start the waymo evaluation...
Number: (pd, 86610) VS. (gt, 16470)
Level 1: 13952, Level2: 2518)
here~~
2024-03-28 19:11:12.728474: I tensorflow/compiler/xla/stream_executor/cuda/cuda_gpu_executor.cc:982] could not open file to read NUMA node: /sys/bus/pci/devices/0000:01:00.0/numa_node
Your kernel may have been built without NUMA support.
2024-03-28 19:11:12.729932: I tensorflow/compiler/xla/stream_executor/cuda/cuda_gpu_executor.cc:982] could not open file to read NUMA node: /sys/bus/pci/devices/0000:01:00.0/numa_node
Your kernel may have been built without NUMA support.
2024-03-28 19:11:12.731113: I tensorflow/compiler/xla/stream_executor/cuda/cuda_gpu_executor.cc:982] could not open file to read NUMA node: /sys/bus/pci/devices/0000:01:00.0/numa_node
Your kernel may have been built without NUMA support.
2024-03-28 19:11:12.732425: I tensorflow/compiler/xla/stream_executor/cuda/cuda_gpu_executor.cc:982] could not open file to read NUMA node: /sys/bus/pci/devices/0000:01:00.0/numa_node
Your kernel may have been built without NUMA support.
2024-03-28 19:11:12.732443: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1722] Could not identify NUMA node of platform GPU id 0, defaulting to 0.  Your kernel may not have been built with NUMA support.
2024-03-28 19:11:12.733590: I tensorflow/compiler/xla/stream_executor/cuda/cuda_gpu_executor.cc:982] could not open file to read NUMA node: /sys/bus/pci/devices/0000:01:00.0/numa_node
Your kernel may have been built without NUMA support.
2024-03-28 19:11:12.733620: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1635] Created device /job:localhost/replica:0/task:0/device:GPU:0 with 12853 MB memory:  -> device: 0, name: NVIDIA GeForce RTX 4070 Ti SUPER, pci bus id: 0000:01:00.0, compute capability: 8.9
here1~~
2024-03-28 19:11:12.737067: I tensorflow/compiler/mlir/mlir_graph_optimization_pass.cc:353] MLIR V1 optimization pass is not enabled
here2~~
2024-03-28 19:11:13.022542: W tensorflow/c/c_api.cc:300] Operation '{name:'Const_22' id:340 op device:{requested: '', assigned: ''} def:{{{node Const_22}} = Const[_has_manual_control_dependencies=true, dtype=DT_FLOAT, value=Tensor<type: float shape: [0] values: >]()}}' was changed by setting attribute after it was run by a session. This mutation will have no effect, and will trigger an error in the future. Either don't modify nodes after running them or create a new session.
here3~~
2024-03-28 19:11:13.149676: I waymo_open_dataset/metrics/ops/detection_metrics_ops.cc:188] Computing detection metrics for 86610 predicted boxes.
2024-03-28 19:11:13.149712: I waymo_open_dataset/metrics/ops/detection_metrics_ops.cc:190] Parsing prediction [86610,7][86610]
2024-03-28 19:11:13.163931: I waymo_open_dataset/metrics/ops/detection_metrics_ops.cc:199] Parsing ground truth [16470,7][16470]
2024-03-28 19:11:15.708036: I waymo_open_dataset/metrics/ops/detection_metrics_ops.cc:252] Done with computing detection metrics.
2024-03-28 19:11:15.730510: I waymo_open_dataset/metrics/ops/detection_metrics_ops.cc:188] Computing detection metrics for 86610 predicted boxes.
2024-03-28 19:11:15.730542: I waymo_open_dataset/metrics/ops/detection_metrics_ops.cc:190] Parsing prediction [86610,7][86610]
2024-03-28 19:11:15.744880: I waymo_open_dataset/metrics/ops/detection_metrics_ops.cc:199] Parsing ground truth [16470,7][16470]
2024-03-28 19:11:18.206481: I waymo_open_dataset/metrics/ops/detection_metrics_ops.cc:252] Done with computing detection metrics.
2024-03-28 19:11:18.229228: I waymo_open_dataset/metrics/ops/detection_metrics_ops.cc:188] Computing detection metrics for 86610 predicted boxes.
2024-03-28 19:11:18.229257: I waymo_open_dataset/metrics/ops/detection_metrics_ops.cc:190] Parsing prediction [86610,7][86610]
2024-03-28 19:11:18.242719: I waymo_open_dataset/metrics/ops/detection_metrics_ops.cc:199] Parsing ground truth [16470,7][16470]
2024-03-28 19:11:20.588735: I waymo_open_dataset/metrics/ops/detection_metrics_ops.cc:252] Done with computing detection metrics.
2024-03-28 19:11:20.612593: I waymo_open_dataset/metrics/ops/detection_metrics_ops.cc:188] Computing detection metrics for 86610 predicted boxes.
[... cut ...]
2024-03-28 19:11:28.133197: I waymo_open_dataset/metrics/ops/detection_metrics_ops.cc:252] Done with computing detection metrics.
2024-03-28 19:11:28.154803: I waymo_open_dataset/metrics/ops/detection_metrics_ops.cc:188] Computing detection metrics for 86610 predicted boxes.
2024-03-28 19:11:28.154838: I waymo_open_dataset/metrics/ops/detection_metrics_ops.cc:190] Parsing prediction [86610,7][86610]
2024-03-28 19:11:28.168891: I waymo_open_dataset/metrics/ops/detection_metrics_ops.cc:199] Parsing ground truth [16470,7][16470]
Illegal instruction`
[image]

closed 2024-05-12 by github-actions[bot]

lichengxi1, 2024-08-28:
hi, I have the same question, have you solved it?

SimonSongg, 2024-10-18:
> hi, I have the same question, have you solved it?

Sorry for late reply, but I found it is due to CPU overclocking. I reset the settings in BIOS and there is no error in the evaluation process.