open-mmlab/OpenPCDet #1596: Issue with KITTI "test set" evaluation turning all 0 values

sinatayebati, 2024-04-06:
I've successfully trained and evaluated several models using "kitti_infos_train.pkl" and "kitti_infos_val.pkl" without any problems. However, when I attempted to test using "kitti_infos_test.pkl," all the returned recall values are showing as 0!

Below, I've provided my configuration and experiment details. I would greatly appreciate it if someone could assist me in troubleshooting this issue. Thank you.

*Openpcdet version*: pcdet+0.6.0

*Testing model*: PVRCNN
- bash scripts/dist_test.sh 2 --cfg_file  cfgs/kitti_models/pv_rcnn.yaml --ckpt ../output/kitti_models/pv_rcnn/default/ckpt/checkpoint_epoch_78.pth

*Evaluation result*:
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
2024-04-06 17:25:30,613   INFO  ==> Loading parameters from checkpoint ../output/kitti_models/pv_rcnn_09_final/default/ckpt/checkpoint_epoch_78.pth to GPU
2024-04-06 17:25:30,695   INFO  ==> Checkpoint trained from version: pcdet+0.6.0+255db8f+pybd06a67
2024-04-06 17:25:30,721   INFO  ==> Done (loaded 367/367)
2024-04-06 17:25:30,740   INFO  *************** EPOCH 78 EVALUATION *****************
eval: 100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 3759/3759 [07:27<00:00,  8.40it/s, recall_0.3=(0, 0) / 0]
2024-04-06 17:32:58,430   INFO  *************** Performance of EPOCH 78 *****************
2024-04-06 17:32:58,430   INFO  Generate label finished(sec_per_example: 0.0595 second).
2024-04-06 17:32:58,430   INFO  recall_roi_0.3: 0.000000
2024-04-06 17:32:58,430   INFO  recall_rcnn_0.3: 0.000000
2024-04-06 17:32:58,430   INFO  recall_roi_0.5: 0.000000
2024-04-06 17:32:58,430   INFO  recall_rcnn_0.5: 0.000000
2024-04-06 17:32:58,430   INFO  recall_roi_0.7: 0.000000
2024-04-06 17:32:58,430   INFO  recall_rcnn_0.7: 0.000000
2024-04-06 17:32:58,441   INFO  Average predicted number of objects(7518 samples): 8.798
2024-04-06 17:32:58,779   INFO  None
2024-04-06 17:32:58,779   INFO  Result is saved to /hdd_10tb/sina/Radial_MAE/output/kitti_models/pv_rcnn/default/eval/epoch_78/test/default
2024-04-06 17:32:58,779   INFO  ****************Evaluation done.*****************

**kitti_dataset.yaml**:

DATASET: 'KittiDataset'
DATA_PATH: '../data/kitti'

POINT_CLOUD_RANGE: [0, -40, -3, 70.4, 40, 1]

DATA_SPLIT: {
    'train': train,
    'test': test
}

INFO_PATH: {
    'train': [kitti_infos_train.pkl],
    'test': [kitti_infos_test.pkl],
}

GET_ITEM_LIST: ["points"]
FOV_POINTS_ONLY: True

DATA_AUGMENTOR:
    DISABLE_AUG_LIST: ['placeholder']
    AUG_CONFIG_LIST:
        - NAME: gt_sampling
          USE_ROAD_PLANE: True
          DB_INFO_PATH:
              - kitti_dbinfos_train.pkl
          PREPARE: {
             filter_by_min_points: ['Car:5', 'Pedestrian:5', 'Cyclist:5'],
             filter_by_difficulty: [-1],
          }

          SAMPLE_GROUPS: ['Car:20','Pedestrian:15', 'Cyclist:15']
          NUM_POINT_FEATURES: 4
          DATABASE_WITH_FAKELIDAR: False
          REMOVE_EXTRA_WIDTH: [0.0, 0.0, 0.0]
          LIMIT_WHOLE_SCENE: True

        - NAME: random_world_flip
          ALONG_AXIS_LIST: ['x']

[... 470 lines cut ...]
torch                         1.10.2
torch-scatter                 2.1.1
torchvision                   0.11.3
tornado                       6.2
tqdm                          4.64.0
traitlets                     5.9.0
transforms3d                  0.3.1
typing_extensions             4.7.1
uri-template                  1.3.0
urllib3                       2.0.7
wandb                         0.16.3
waymo-open-dataset-tf-2-5-0   1.4.1
wcwidth                       0.2.10
webcolors                     1.13
webencodings                  0.5.1
websocket-client              1.6.1
Werkzeug                      2.2.3
wheel                         0.42.0
widgetsnbextension            4.0.10
wrapt                         1.12.1
xmltodict                     0.13.0
y-py                          0.6.2
yacs                          0.1.8
ypy-websocket                 0.8.4
zipp                          3.15.0

Dai-Yu-Chen, 2024-04-11:
Hi, I have the same problem. Have you solved this problem?

sinatayebati, 2024-04-11:
> Hi, I have the same problem. Have you solved this problem?

Unfortunately not!

calcualatexzy, 2024-05-14:
It is very annoying that I saw the same problem in "issues" several times and the team keeps on closing the issues due to 30 days stale, so is there any solution?

josyulavt, 2024-05-17:
I have the same problem during training the loss is shown to be 0.2, I wanted to see the test results and everything is 0. 

Any help is much appreciated

closed 2024-07-01 by github-actions[bot]

llz666ha, 2025-06-04:
hello,anyone solved? Is it just able to use "val"?how about "test"?

ImenMabroukIMSR, 2025-07-01:
@sinatayebati Hi, I have the same problem. Have you solved this problem?

Sun-Cold, 2025-09-29:
GT labels on test set are not public. The only way to evaluate kitti's test set is upload your results to kitti website.