open-mmlab/OpenPCDet #1619: Low Acc on 3D task, in Custom datasets.

Mike-7777777, 2024-05-22:
Thank you for your excellent work. 

I'm trying to use OpenPCDet's customized dataset feature, but I'm finding that my models are performing particularly poorly for BEV and 3D tasks, does anyone have any idea what this might be? Or how I should troubleshoot the problem?

Another question is that when I check (visualize) custom_infos_train.pkl, I find that some labels that are outside the range of the point cloud are not removed, is this normal?

I am using the TUMTraf dataset. You can check my custom dataset and model config files in my fork if you want.

My log file:

`
2024-05-20 22:53:41,737   INFO  **********************Start logging**********************
2024-05-20 22:53:41,738   INFO  CUDA_VISIBLE_DEVICES=0
2024-05-20 22:53:41,738   INFO  Training with a single process
2024-05-20 22:53:41,738   INFO  cfg_file         /user/<user.name>/<user.name2>/OpenPCDet/tools/cfgs/custom_models/pps.yaml
2024-05-20 22:53:41,738   INFO  batch_size       4
2024-05-20 22:53:41,738   INFO  epochs           15
2024-05-20 22:53:41,738   INFO  workers          4
2024-05-20 22:53:41,738   INFO  extra_tag        default
2024-05-20 22:53:41,738   INFO  ckpt             None
2024-05-20 22:53:41,738   INFO  pretrained_model None
2024-05-20 22:53:41,738   INFO  launcher         none
2024-05-20 22:53:41,738   INFO  tcp_port         18888
2024-05-20 22:53:41,738   INFO  sync_bn          False
2024-05-20 22:53:41,738   INFO  fix_random_seed  False
2024-05-20 22:53:41,738   INFO  ckpt_save_interval 1
2024-05-20 22:53:41,738   INFO  local_rank       None
2024-05-20 22:53:41,738   INFO  max_ckpt_save_num 30
2024-05-20 22:53:41,738   INFO  merge_all_iters_to_one_epoch False
2024-05-20 22:53:41,738   INFO  set_cfgs         None
2024-05-20 22:53:41,738   INFO  max_waiting_mins 0
2024-05-20 22:53:41,738   INFO  start_epoch      0
2024-05-20 22:53:41,738   INFO  num_epochs_to_eval 0
2024-05-20 22:53:41,738   INFO  save_to_file     False
2024-05-20 22:53:41,738   INFO  use_tqdm_to_record False
2024-05-20 22:53:41,738   INFO  logger_iter_interval 50
2024-05-20 22:53:41,739   INFO  ckpt_save_time_interval 300
2024-05-20 22:53:41,739   INFO  wo_gpu_stat      False
2024-05-20 22:53:41,739   INFO  use_amp          False
2024-05-20 22:53:41,739   INFO  cfg.ROOT_DIR: /home/uni08/hpc/<user.name>/u11423/OpenPCDet
2024-05-20 22:53:41,739   INFO  cfg.LOCAL_RANK: 0
2024-05-20 22:53:41,739   INFO  cfg.CLASS_NAMES: ['CAR', 'PEDESTRIAN', 'BICYCLE']
2024-05-20 22:53:41,739   INFO  ----------- DATA_CONFIG -----------
2024-05-20 22:53:41,739   INFO  cfg.DATA_CONFIG.DATASET: CustomDataset
2024-05-20 22:53:41,739   INFO  cfg.DATA_CONFIG.DATA_PATH: ../data/custom
2024-05-20 22:53:41,739   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [0.0, -180.0, -10.0, 153.6, 137.44, 0.0]
2024-05-20 22:53:41,739   INFO  ----------- MAP_CLASS_TO_KITTI -----------
2024-05-20 22:53:41,739   INFO  cfg.DATA_CONFIG.MAP_CLASS_TO_KITTI.CAR: Car
2024-05-20 22:53:41,739   INFO  cfg.DATA_CONFIG.MAP_CLASS_TO_KITTI.VAN: Van
2024-05-20 22:53:41,739   INFO  cfg.DATA_CONFIG.MAP_CLASS_TO_KITTI.TRUCK: Truck
2024-05-20 22:53:41,739   INFO  cfg.DATA_CONFIG.MAP_CLASS_TO_KITTI.PEDESTRIAN: Pedestrian
2024-05-20 22:53:41,739   INFO  cfg.DATA_CONFIG.MAP_CLASS_TO_KITTI.BICYCLE: Cyclist
2024-05-20 22:53:41,739   INFO  cfg.DATA_CONFIG.MAP_CLASS_TO_KITTI.MOTORCYCLE: Misc
2024-05-20 22:53:41,739   INFO  cfg.DATA_CONFIG.MAP_CLASS_TO_KITTI.TRAILER: Misc
2024-05-20 22:53:41,739   INFO  cfg.DATA_CONFIG.MAP_CLASS_TO_KITTI.BUS: Misc
2024-05-20 22:53:41,739   INFO  cfg.DATA_CONFIG.MAP_CLASS_TO_KITTI.EMERGENCY_VEHICLE: Misc
2024-05-20 22:53:41,739   INFO  ----------- DATA_SPLIT -----------
2024-05-20 22:53:41,739   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: train
2024-05-20 22:53:41,739   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: val
2024-05-20 22:53:41,739   INFO  ----------- INFO_PATH -----------
2024-05-20 22:53:41,739   INFO  cfg.DATA_CONFIG.INFO_PATH.train: ['custom_infos_train.pkl']
2024-05-20 22:53:41,739   INFO  cfg.DATA_CONFIG.INFO_PATH.test: ['custom_infos_val.pkl']
2024-05-20 22:53:41,739   INFO  ----------- POINT_FEATURE_ENCODING -----------
2024-05-20 22:53:41,739   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
2024-05-20 22:53:41,739   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.used_feature_list: ['x', 'y', 'z', 'intensity']
2024-05-20 22:53:41,739   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.src_feature_list: ['x', 'y', 'z', 'intensity']
2024-05-20 22:53:41,739   INFO  ----------- DATA_AUGMENTOR -----------
2024-05-20 22:53:41,739   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder']
2024-05-20 22:53:41,739   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'gt_sampling', 'USE_ROAD_PLANE': False, 'DB_INFO_PATH': ['custom_dbinfos_train.pkl'], 'PREPARE': {'filter_by_min_points': ['CAR:5', 'PEDESTRIAN:5', 'BICYCLE:5']}, 'SAMPLE_GROUPS': ['CAR:15', 'PEDESTRIAN:15', 'BICYCLE:15'], 'NUM_POINT_FEATURES': 4, 'DATABASE_WITH_FAKELIDAR': False, 'REMOVE_EXTRA_WIDTH': [0.0, 0.0, 0.0], 'LIMIT_WHOLE_SCENE': False}, {'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.78539816, 0.78539816]}, {'NAME': 'random_world_scaling', 'WORLD_SCALE_RANGE': [0.95, 1.05]}]
2024-05-20 22:53:41,740   INFO  cfg.DATA_CONFIG.DATA_PROCESSOR: [{'NAME': 'mask_points_and_boxes_outside_range', 'REMOVE_OUTSIDE_BOXES': True}, {'NAME': 'shuffle_points', 'SHUFFLE_ENABLED': {'train': True, 'test': False}}, {'NAME': 'transform_points_to_voxels', 'VOXEL_SIZE': [0.64, 0.64, 10], 'MAX_POINTS_PER_VOXEL': 32, 'MAX_NUMBER_OF_VOXELS': {'train': 230000, 'test': 230000}}]
[... 262 lines cut ...]
bbox AP:53.1667, 53.1667, 53.1667
bev  AP:20.6734, 20.6734, 20.6734
3d   AP:19.6189, 19.6189, 19.6189
aos  AP:42.44, 42.44, 42.44
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:22.1108, 22.1108, 22.1108
bev  AP:2.0695, 2.0695, 2.0695
3d   AP:0.3497, 0.3497, 0.3497
aos  AP:21.56, 21.56, 21.56
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:20.5448, 20.5448, 20.5448
bev  AP:1.1436, 1.1436, 1.1436
3d   AP:0.1923, 0.1923, 0.1923
aos  AP:19.94, 19.94, 19.94
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:22.1108, 22.1108, 22.1108
bev  AP:3.9044, 3.9044, 3.9044
3d   AP:2.0695, 2.0695, 2.0695
aos  AP:21.56, 21.56, 21.56
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:20.5448, 20.5448, 20.5448
bev  AP:2.2639, 2.2639, 2.2639
3d   AP:1.1436, 1.1436, 1.1436
aos  AP:19.94, 19.94, 19.94
`

closed 2024-07-05 by github-actions[bot]

YouthZCC, 2024-08-04:
have u solved the problem? If yes,how?

NurbekkWARTO, 2024-09-12:
is there any update?