open-mmlab/OpenPCDet #1624: Error when running demo.py

AlexandreDiPiazza, 2024-06-06:
When running demo.py, I run into the error: 

----

[Open3D WARNING] Failed to initialize GLFW
Traceback (most recent call last):
  File "demo.py", line 124, in <module>
    main()
  File "demo.py", line 111, in main
    V.draw_scenes(
  File "/home/ubuntu/alex/OpenPCDet/tools/visual_utils/open3d_vis_utils.py", line 49, in draw_scenes
    vis.get_render_option().point_size = 1.0
AttributeError: 'NoneType' object has no attribute 'point_size'

----

I saw the error mentionned before, but couldn't figure out to fix it. Does someone have a solution?

Hberto, 2024-06-13:
I have the same issue when training with pv rcnn with the custom dataset and running it on Google Collab. I'll try to run it again on a pc with a gpu. I guess you already saw these discussions:
https://github.com/open-mmlab/mmdetection3d/issues/344
https://github.com/open-mmlab/OpenPCDet/issues/748

AlexandreDiPiazza, 2024-06-13:
Hey,

I add this error when using a VM. When using my personal computer I could use demo.py without errors.

closed 2024-06-14 by AlexandreDiPiazza