open-mmlab/OpenPCDet #1627: Abnormal demo.py result using nuscenses dataset pre-trained models

yaobaishen, 2024-06-12:
This issue is related to [issue 1341](https://github.com/open-mmlab/OpenPCDet/issues/1341), but here we track a different problem.
After modify the demo.py a bit in order to run the nuscenses dataset pre-trained model, I get weird inference result.
The modification is below, because the nuscenses lidar data has 5 dimensions: ['x', 'y', 'z', 'intensity', 'timestamp'].
```
    def __getitem__(self, index):
        if self.ext == '.bin':
            if self.is_nuscense_data:
                points = np.fromfile(self.sample_file_list[index],
                                     dtype=np.float32).reshape(-1, 5)
            else:
                points = np.fromfile(self.sample_file_list[index],
                                     dtype=np.float32).reshape(-1, 4)
```
And the command line to test CenterPoint model with demo.py. Note: I have added a --nuscense_data flag, and test with a xxx.pcd.bin file from nuscenes_mini.
```
python demo.py --cfg_file cfgs/nuscenes_models/cbgs_dyn_pp_centerpoint.yaml --ckpt ../model_zoo/nuscenes_models/cbgs_pp_centerpoint_nds6070.pth --data_path dataset/nuscenes_mini/samples/LIDAR_TOP/n015-2018-11-21-19-38-26+0800__LIDAR_TOP__1542801004447480.pcd.bin --nuscense_data
```
Here is the inference result, even after I set a score threshold 0.3, there are many abnormal detection boxes:

[image]

Could anyone share some insights about how to use the nuscenses dataset pre-trained models? Thanks a lot.

yaobaishen, 2024-06-12:
I also try the PointPillar model download from the main-page, the inference result still looks strange.
command line:
```
python demo.py --cfg_file cfgs/nuscenes_models/cbgs_pp_multihead.yaml --ckpt ../model_zoo/nuscenes_models/pp_multihead_nds5823_updated.pth --data_path dataset/nuscenes_mini/samples/LIDAR_TOP/n015-2018-11-21-19-38-26+0800__LIDAR_TOP__1542801004447480.pcd.bin --nuscense_data
```
visulization:
[image]

closed 2024-07-28 by github-actions[bot]

muvanpersie, 2025-03-26:
Hi, have you solve the probelm yet? I find that the format of  nuscenes lidar "bin" file is [x, y, z, intensity, ring_id], the last channel is ring_id but not timestamp. So I think we should drop the last "ring_id" channel and concat with the right "timestamp".

yaobaishen, 2025-03-27:
@muvanpersie thanks for update,  so how about your result looks like? I haven't look into this project for a long while.

muvanpersie, 2025-03-27:
> [@muvanpersie](https://github.com/muvanpersie) thanks for update, so how about your result looks like? I haven't look into this project for a long while.

Much better but with a few false alarms after simply replace the **DemoDataset** by **NusencesDataset**.  And for 'bin' files, we should transform the other 9 sweeps frame to the 1 sample  frame, with the time difference as the fifth "timestamp" channel. 

[image]

[image]

yaobaishen, 2025-04-19:
Sorry for late update. Thanks for @muvanpersie 's tips, after replace the DemoDataset by NusencesDataset, and create the nuscenes_infos_10sweeps_train.pkl / nuscenes_infos_10sweeps_val.pkl following the [official guide](https://github.com/open-mmlab/OpenPCDet/blob/master/docs/GETTING_STARTED.md), the visualization of demo.py looks much better now.

[image]

yaobaishen, 2025-06-03:
@HHX0607 You only need to change the **DemoDataset** to **NuScenesDataset**, just like what muvanpersie paste in
https://github.com/open-mmlab/OpenPCDet/issues/1627#issuecomment-2757329839
Then the demo.py will process the data files specified in cfg.DATA_CONFIG, you don't need to specify the bin files in command line. I use the demo.py like this:
`
python demo.py --cfg_file cfgs/nuscenes_models/cbgs_second_multihead.yaml  --ckpt ../model_zoo/nuscenes_models/cbgs_second_multihead_nds6229_updated.pth
`