open-mmlab/OpenPCDet #1631: 'EasyDict' object has no attribute 'BACKUP_DB_INFO'

Itay-Krispin, 2024-07-04:
Hi,

I have worked with the framework before and I remeber encountering this problem below but I can't recall how did I solve it :(
I have created the dbinfos successully and it seems that all the paths are correct (see screen shots below) so I am not sure what's the issue.
Does someone know?

 2024-07-04 12:43:20,294   INFO  ----------- Create dataloader & network & optimizer -----------
Traceback (most recent call last):
  File "/home/tauproj1/Innoviz_Project/OpenPCDet-master_New_copy/tools/train.py", line 277, in <module>
    main()
  File "/home/tauproj1/Innoviz_Project/OpenPCDet-master_New_copy/tools/train.py", line 145, in main
    train_set, train_loader, train_sampler = build_dataloader(
                                             ^^^^^^^^^^^^^^^^^
  File "/home/tauproj1/Innoviz_Project/OpenPCDet-master_New_copy/tools/../pcdet/datasets/__init__.py", line 69, in build_dataloader
    dataset = __all__[dataset_cfg.DATASET](
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tauproj1/Innoviz_Project/OpenPCDet-master_New_copy/tools/../pcdet/datasets/custom/custom_dataset.py", line 21, in __init__
    super().__init__(
  File "/home/tauproj1/Innoviz_Project/OpenPCDet-master_New_copy/tools/../pcdet/datasets/dataset.py", line 31, in __init__
    self.data_augmentor = DataAugmentor(
                          ^^^^^^^^^^^^^^
  File "/home/tauproj1/Innoviz_Project/OpenPCDet-master_New_copy/tools/../pcdet/datasets/augmentor/data_augmentor.py", line 24, in __init__
    cur_augmentor = getattr(self, cur_cfg.NAME)(config=cur_cfg)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tauproj1/Innoviz_Project/OpenPCDet-master_New_copy/tools/../pcdet/datasets/augmentor/data_augmentor.py", line 40, in gt_sampling
    db_sampler = database_sampler.DataBaseSampler(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tauproj1/Innoviz_Project/OpenPCDet-master_New_copy/tools/../pcdet/datasets/augmentor/database_sampler.py", line 35, in __init__
    sampler_cfg.DB_INFO_PATH[0] = sampler_cfg.BACKUP_DB_INFO['DB_INFO_PATH']
                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'EasyDict' object has no attribute 'BACKUP_DB_INFO'

[image]

[image]

Itay-Krispin, 2024-07-10:
Hi, 
Any ideas?

Zakaria1405, 2024-07-22:
are you working with mini nuscenes_dataset?
if yes go to: OpenPCDet/tools/cfgs/dataset_configs/nuscenes_dataset.yaml
and change the version to : VERSION: 'v1.0-mini'
Otherwise always check the direktory to your .pkl file

ItayKrispin, 2024-07-22:
> are you working with mini nuscenes_dataset? if yes go to: OpenPCDet/tools/cfgs/dataset_configs/nuscenes_dataset.yaml and change the version to : VERSION: 'v1.0-mini' Otherwise always check the direktory to your .pkl file

Thanks for replying. No I am working with a custom dataset but I figured out the problem :)

allenwgb, 2024-08-15:
> > are you working with mini nuscenes_dataset? if yes go to: OpenPCDet/tools/cfgs/dataset_configs/nuscenes_dataset.yaml and change the version to : VERSION: 'v1.0-mini' Otherwise always check the direktory to your .pkl file
> 
> Thanks for replying. No I am working with a custom dataset but I figured out the problem :)
> 
> The issue can be closed now.

I enconted the same problem, could you share how do you fix it please?

yyxmoon123, 2024-09-25:
how procees

matren12138, 2024-10-06:
I had the same problem before as well, for me the reason seemed to be incomplete dataset

Petros626, 2024-10-12:
@ItayKrispin when you solved it, post a solution for others, this is a community.

ItayKrispin, 2024-10-12:
Sorry guys haven’t seen the comments until now. 
In my case it turned out that my data folder wasn’t under the right parent directory… nothing special to contribute just an overlook by me.

Petros626, 2024-10-12:
some additional hint, when using the debugger of vscode, make sure that the paths are correctly set. In this image you can see the false path, using a custom dataset.

[image]

kistothm, 2024-11-14:
I had the same issue. I solved it by modifying the "DATA_PATH" in the kitti dataset config file. I set it from a relative path to the absolute path of the dataset.

closed 2024-12-30 by github-actions[bot]