open-mmlab/OpenPCDet #1632: What should we set the overlap values for custom datasets?

silva21d, 2024-07-05:
I'm trying testing a model with my custom dataset. The number of classes is 10.

When I executed `tools/test.py`, a error occurred in get_official_eval_result function in `pcdet/datasets/kitti/kitti_object_eval_python/eval.py`

The details is like below.
```
  File "xxxx/datasets/kitti/kitti_object_eval_python/eval.py", line xxx, in get_official_eval_result
    min_overlaps = min_overlaps[:, :, current_classes]
IndexError: index 6 is out of bounds for axis 2 with size 6
```
min_overlaps consists of overlap_0_7 and overlap_0_5.
The overlaps are like below.

```
def get_official_eval_result(gt_annos, dt_annos, current_classes, PR_detail_dict=None):
    overlap_0_7 = np.array([[0.7, 0.5, 0.5, 0.7,
                             0.5, 0.7], [0.7, 0.5, 0.5, 0.7, 0.5, 0.7],
                            [0.7, 0.5, 0.5, 0.7, 0.5, 0.7]])
    overlap_0_5 = np.array([[0.7, 0.5, 0.5, 0.7,
                             0.5, 0.5], [0.5, 0.25, 0.25, 0.5, 0.25, 0.5],
                            [0.5, 0.25, 0.25, 0.5, 0.25, 0.5]])
    min_overlaps = np.stack([overlap_0_7, overlap_0_5], axis=0)  # [2, 3, 5]
```

I don't understand that each value of cols and rows of overlap_0_5 and overlap_0_7 for my custom dataset's classes.

Why are not all values of overlap_0_7  0.7? And also why are not all values of overlap_0_5  0.5.?
Please tell me what values should I set on the overlaps for my custom dataset classes (total 10 classes).

yinheyanxian, 2024-07-08:
just try
overlap_0_7 = np.array([[0.7,0.7,0.7,0.7,0.7,0.7,0.7,0.7,0.7,0.7], 
                                      [0.7,0.7,0.7,0.7,0.7,0.7,0.7,0.7,0.7,0.7],
                                      [0.7,0.7,0.7,0.7,0.7,0.7,0.7,0.7,0.7,0.7]])
overlap_0_5 = np.array([[0.5,0.5,0.5,0.5,0.5,0.5,0.5,0.5,0.5,0.5], 
                                      [0.5,0.5,0.5,0.5,0.5,0.5,0.5,0.5,0.5,0.5], 
                                      [0.5,0.5,0.5,0.5,0.5,0.5,0.5,0.5,0.5,0.5]])

silva21d, 2024-07-10:
@yinheyanxian 

Thank you for your answer.
Your advice is good. The error doesn't occur.

I have another question. Why official source code take the numerical value for overlap besides 0.7 and 0.5?
The variable names doesn't suit with the numerical values.
Is it fine tuning for each classes?

yinheyanxian, 2024-07-11:
> @yinheyanxian
> 
> Thank you for your answer. Your advice is good. The error doesn't occur.
> 
> I have another question. Why official source code take the numerical value for overlap besides 0.7 and 0.5? The variable names doesn't suit with the numerical values. Is it fine tuning for each classes?

What do you mean by 'the numerical value'? What line of code is this?

liuyuansheng624, 2024-07-15:
I can train and test on openpcdet using custom point cloud datasets
weichat13731082126

closed 2024-08-29 by github-actions[bot]