open-mmlab/OpenPCDet #1634: `pillar_features` IndexError: too many indices for tensor of dimension 1

Oh-JongJin, 2024-07-09:
```bash
(openpcdet) ves@ves:~/OpenPCDet/tools$ python demo.py --cfg_file cfgs/kitti_models/pointpillar.yaml --ckpt pointpillar_7728.pth --data_path /media/ves/Z/bag_ansan/pcd_out/npy_list/1718073773.002808809.npy 
2024-07-09 13:41:48,997   INFO  -----------------Quick Demo of OpenPCDet-------------------------
2024-07-09 13:41:48,997   INFO  Total number of samples: 	1
/home/ves/anaconda3/envs/openpcdet/lib/python3.9/site-packages/torch/functional.py:504: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at /opt/conda/conda-bld/pytorch_1702400441250/work/aten/src/ATen/native/TensorShape.cpp:3526.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
2024-07-09 13:41:49,148   INFO  ==> Loading parameters from checkpoint pointpillar_7728.pth to CPU
2024-07-09 13:41:49,165   INFO  ==> Done (loaded 127/127)
2024-07-09 13:41:49,181   INFO  Visualized sample index: 	1
Traceback (most recent call last):
  File "/home/ves/OpenPCDet/tools/demo.py", line 138, in <module>
    main()
  File "/home/ves/OpenPCDet/tools/demo.py", line 124, in main
    pred_dicts, _ = model.forward(data_dict)
  File "/home/ves/OpenPCDet/pcdet/models/detectors/pointpillar.py", line 12, in forward
    batch_dict = cur_module(batch_dict)
  File "/home/ves/anaconda3/envs/openpcdet/lib/python3.9/site-packages/torch/nn/modules/module.py", line 1518, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/ves/anaconda3/envs/openpcdet/lib/python3.9/site-packages/torch/nn/modules/module.py", line 1527, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/ves/OpenPCDet/pcdet/models/backbones_2d/map_to_bev/pointpillar_scatter.py", line 41, in forward
    pillars = pillar_features[batch_mask, :]
IndexError: too many indices for tensor of dimension 1
```
[my_npy_file.zip](https://github.com/user-attachments/files/16138192/my_npy_file.zip)

The **`pillar_features`** of the npy file that works for 3D object detection is a 2D array, but the **`pillar_features`** of my npy file is output as a 1D array.
```python
class PointPillarScatter(nn.Module):
    def __init__(self, model_cfg, grid_size, **kwargs):
        super().__init__()
        self.model_cfg = model_cfg
        self.num_bev_features = self.model_cfg.NUM_BEV_FEATURES
        self.nx, self.ny, self.nz = grid_size
        assert self.nz == 1

    def forward(self, batch_dict, **kwargs):
        # print(batch_dict)
        pillar_features, coords = batch_dict['pillar_features'], batch_dict['voxel_coords']
        batch_spatial_features = []
        batch_size = coords[:, 0].max().int().item() + 
        
        # By using this code, the point cloud is displayed, but the 3D boxes are not.
        # if len(pillar_features.shape) == 1:
        #     pillar_features = pillar_features.unsqueeze(0)

        for batch_idx in range(batch_size):
            spatial_feature = torch.zeros(
                self.num_bev_features,
                self.nz * self.nx * self.ny,
                dtype=pillar_features.dtype,
                device=pillar_features.device)

            batch_mask = coords[:, 0] == batch_idx
            this_coords = coords[batch_mask, :]
            indices = this_coords[:, 1] + this_coords[:, 2] * self.nx + this_coords[:, 3]
            indices = indices.type(torch.long)
            
            print('*****')
            print(f'pillar_features: \n{pillar_features}\npillar_features.shape: {pillar_features.shape}')
            print('-----------------------------')
            print(f'batch_mask: \n{batch_mask}\nbatch_mask.shape: {batch_mask.shape}')
            print('*****')
```
->
```bash
python demo.py --cfg_file cfgs/kitti_models/pointpillar.yaml --ckpt pointpillar_7728.pth --data_path pointcloud_test.npy
``` 
```
pillar_features: 
tensor([0.0000e+00, 1.2826e+32, 4.2726e+31, 1.5496e+31, 2.6135e+30, 8.7043e-01,
        2.6156e-01, 1.0135e+29, 5.3591e-01, 2.0796e+29, 1.7418e+31, 0.0000e+00,
        6.4862e+28, 0.0000e+00, 4.0649e-01, 2.6790e+31, 2.8577e+32, 5.4466e+29,
        6.6605e+31, 3.5346e+31, 0.0000e+00, 0.0000e+00, 1.8015e-01, 4.4836e+30,
        1.3956e+32, 1.2039e+29, 3.6126e-03, 0.0000e+00, 0.0000e+00, 2.6504e+30,
        0.0000e+00, 3.7262e-01, 1.6756e-01, 4.0609e+29, 1.6201e+31, 2.0788e+31,
        2.7670e+00, 7.9919e+30, 6.3981e+31, 0.0000e+00, 2.0904e+00, 4.9841e+30,
        1.7689e+31, 0.0000e+00, 7.4313e+30, 0.0000e+00, 0.0000e+00, 3.7969e+31,
        8.6608e-02, 1.2728e+30, 4.5646e-01, 1.9545e-01, 0.0000e+00, 2.8419e+32,
        1.6905e+29, 1.3752e+32, 3.9255e+31, 7.6501e+29, 2.7532e+32, 6.4052e+31,
        0.0000e+00, 3.3674e+30, 2.3766e-02, 4.8696e+31], device='cuda:0')
pillar_features.shape: torch.Size([64])
-----------------------------
batch_mask: 
tensor([True], device='cuda:0')
batch_mask.shape: torch.Size([1])
```

What could be the reason for this, and is there any solution?

closed 2024-08-24 by github-actions[bot]

Hberto, 2024-09-06:
@Oh-JongJin Same issue here, any further advancements or solutions on this issue?

TianchaoHuo, 2025-08-13:
@Oh-JongJin @Hberto Same issue  I meet. Have any suggestion?