open-mmlab/OpenPCDet #1642: Floating Point Exception

jyothsna-phd22, 2024-08-01:
Hi,

While I was training any model on kitti dataset, I am getting floating point exception at epoch 0 and terminated training.

Kindly help me....

beijbom, 2024-08-07:
Same here. I was running `cfgs/nuscenes_models/cbgs_pp_multihead.yaml` on the nuScenes mini dataset. @jyothsna-phd22 : were you able to resolve the issue?

beijbom, 2024-08-07:
Tracked down the offending call. It's `tv.from_numpy(points)` on this line: https://github.com/open-mmlab/OpenPCDet/blob/master/pcdet/datasets/processor/data_processor.py#L55.

jyothsna-phd22, 2024-08-07:
Where should we change in this line??

beijbom, 2024-08-07:
I don't yet know how to fix it. I was just sharing that it seems to be the offending line.

beijbom, 2024-08-07:
On digging further, it seems this is due to a mismatch between the binaries of the spconv library and the rest of the environment. I got it to run using the dockerfile in this PR: https://github.com/open-mmlab/OpenPCDet/pull/1648

Sun-Cold, 2024-09-10:
Same issue here，this is my env info：
python    3.10.14
CUDA     12.1
numpy    2.0.1
spconv-cu120    2.3.6
Here is my solution for reference: downgrade numpy version to 1.26.4
`conda install numpy=1.26.4`

closed 2024-10-25 by github-actions[bot]

likecaulfield, 2025-01-28:
> Same issue here，this is my env info： python 3.10.14 CUDA 12.1 numpy 2.0.1 spconv-cu120 2.3.6 Here is my solution for reference: downgrade numpy version to 1.26.4 `conda install numpy=1.26.4`

It worked for me as well thank you!