open-mmlab/OpenPCDet #1665: Confusion matrix

jyothsna-phd22, 2024-09-13:
Hi,
Is there code base for confusion matrix???

Petros626, 2024-09-18:
Did the codebase contained any methods for it?

binshaea, 2024-09-30:
Hi, I have the same question and I already went through the code
I am using PV-RCNN model for training with custom dataset and I could not find a code that computes confusion matrix or model accuracy. if anyone can confirm this or is there a similar code for these metrics in the other models provided by OpenPCDet?

Petros626, 2024-09-30:
I mean my question was more like, did you searched all the code for a hint. I couldn't find anything that would fit a confusion matrix.

[image]

Petros626, 2024-10-15:
@binshaea when you didn't found anything in the code base, please close this issue.

closed 2024-11-29 by github-actions[bot]