open-mmlab/OpenPCDet #1666: demo.py: error: unrecognized arguments

little-win, 2024-09-16:
when i want to use the demo to deal with my data, it failed
root@autodl-container-b0a041be91-e7140047:~/OpenPCDet/tools# python3 demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml \ --ckpt pv_rcnn_8369.pth \ --data_path 000030g.npy
usage: demo.py [-h] [--cfg_file CFG_FILE] [--data_path DATA_PATH] [--ckpt CKPT] [--ext EXT]
demo.py: error: unrecognized arguments:  --ckpt pv_rcnn_8369.pth  --data_path 000030g.npy
and the things are in the tools
[image]
[image]

little-win, 2024-09-16:
some ones here？

Petros626, 2024-09-17:
I think this repo isn't maintained anmyore. Everything must be figured out on our own and with the old issues.

little-win, 2024-09-18:
python3 demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml --ckpt pv_rcnn_8369.pth --data_path 000030g.npy
it should remove the '\' between the path, it work, but i need to solve other question.

little-win, 2024-09-18:
done

closed 2024-09-18 by little-win