open-mmlab/OpenPCDet #1668: error:python setup.py develop

zscya, 2024-09-24:
But the cuda I installed is 11.1 version, including the setting of environment variables is also 11.1 version, I do not know why it can not find cuda12.2 error
[image]
[image]
[image]
[image]
[image]
[image]

zscya, 2024-09-24:
can you help me?thank you,you will get many papers in you future

zscya, 2024-10-11:
thank you,i have solve it

Petros626, 2024-10-11:
@zscya no problem, please close the issue.

closed 2024-11-25 by github-actions[bot]