open-mmlab/OpenPCDet #1669: Kitti Dataset Split -Reg

jyothsna-phd22, 2024-09-25:
Hi,
I have made custom split the training folder of kitti dataset with 80% for training, 10% for validation and 10% for testing. I have created pkl files with the above split. I am able to train the model, find validation accuracy of the model from the validation split, visualize the model predictions on test set. But when I am trying to find results on test set, it is showing recall values as 0 and no AP results. Why is it happening??
Training: 5984, validation: 748, testing: 749 samples.

I have used my custom kitti_infos_test.py, in dataset_configs/kitti_dataset.yaml file was modified with data split as test: test, infopath as kitti_infos_test.pkl.

Is it possible to find results on custom test split? If so, what is the issue?

10% test set data I have put in testing folder i.e., 749 files are placed in each of the sub folders in testing folder. 

Kindly help me what should be done?

[image]

Petros626, 2024-09-27:
This repo seems not to be maintained anymore, you must search through the issues

Marcelomm103, 2024-10-02:
I’ve had this issue using kitti and test program as well. For me the solution was to use just the validation data, not test data

shingszelam, 2024-10-14:
Because the code part needs to be judged by the gt value and the predicted value, but the gt value of the test set is not released by the kitti official website, you need to apply for submission and verification after training trainval.

closed 2024-11-28 by github-actions[bot]