open-mmlab/OpenPCDet #1671: Compile without GPU

XiangyPeng, 2024-10-14:
Hi, thank you for the great work.

Is that possible to compile "OpenPCDet" without GPU？
Our GPU will automatically cut down the script if the utilization rate of the GPU is lower than 5% for 5min. But the compile process takes longer than 5min. Therefore, I would like to know if it is possible to compile the "setup.py" without GPU.

Thank you in advance!

Petros626, 2024-10-15:
I think you have to read through the old issues, somebody did it with CPU only I guess, but I'm not sure.

XiangyPeng, 2024-10-15:
> I think you have to read through the old issues. Somebody did it with CPU only, I guess, but I'm not sure.

Hi, thank you for your reply. However, I checked the old issues, and I think someone trained it by CPU. But I am wondering if it is doable to compile it by CPU. For me, the problem would be during compiling, the GPU usage rate is too low, which is not allowed.

Petros626, 2024-10-15:
Then it seems there is no other way, you must bypass the low GPU usage rate or you can't install OpenPCDet properly.

XiangyPeng, 2024-10-15:
Okay. Thank you for your suggestion.

Petros626, 2024-10-16:
@XiangyPeng you can reopen the issue later, but for now it seems you have to compile it with GPU.

closed 2024-11-30 by github-actions[bot]