open-mmlab/OpenPCDet #1672: How do I submit a model trained with pcdet to Kitti for evaluation?

shingszelam, 2024-10-14:
I trained a model and wanted to see the results on the test set, but I couldn't find the corresponding file to submit to Kitti. Can someone help me?

Petros626, 2024-10-15:
some users mentioned that in the issues.

shingszelam, 2024-10-16:
> some users mentioned that in the issues.

Is there a solution? I didn't see one.

Petros626, 2024-10-16:
I've read it in the issues with keywords(test, eval, évaluation etc.) you will find it. I remember that user OrangeSodaHub wrote a custom evaluation function, but I never used it.

shingszelam, 2024-10-16:
> (

Ok, thank you!!

shingszelam, 2024-10-16:
> I've read it in the issues with keywords(test, eval, évaluation etc.) you will find it. I remember that user OrangeSodaHub wrote a custom evaluation function, but I never used it.

ok,thank you!!

Petros626, 2024-10-16:
@shingszelam please close this issue then.

closed 2024-10-16 by shingszelam