open-mmlab/OpenPCDet #1684: train error

pedestrain123, 2024-11-15:
训练自己的数据集遇到这个问题KeyError: 'difficulty'，请问怎么解决？

Petros626, 2024-11-25:
@pedestrain123 seems, that you got no category of difficulty in your labels.

closed 2025-01-09 by github-actions[bot]

liyufan, 2025-05-06:
Just remove the `DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST.gt_sampling.PREPARE.filter_by_difficulty` key from your model config. This is my `voxel_rcnn_pedestrian.yaml`:
```yaml
CLASS_NAMES: ['Pedestrian']

DATA_CONFIG:
    _BASE_CONFIG_: cfgs/dataset_configs/custom_dataset.yaml
    DATA_AUGMENTOR:
        DISABLE_AUG_LIST: ['placeholder']
        AUG_CONFIG_LIST:
            - NAME: gt_sampling
              USE_ROAD_PLANE: False
              DB_INFO_PATH:
                  - custom_dbinfos_train.pkl
              PREPARE: {
                 filter_by_min_points: ['Pedestrian:5'],
                 # here, delete `filter_by_difficulty`
              }

              SAMPLE_GROUPS: ['Pedestrian:15']
              NUM_POINT_FEATURES: 3
              DATABASE_WITH_FAKELIDAR: False
              REMOVE_EXTRA_WIDTH: [0.0, 0.0, 0.0]
              LIMIT_WHOLE_SCENE: False

            - NAME: random_world_flip
              ALONG_AXIS_LIST: ['x']

            - NAME: random_world_rotation
              WORLD_ROT_ANGLE: [-0.78539816, 0.78539816]

            - NAME: random_world_scaling
              WORLD_SCALE_RANGE: [0.95, 1.05]
```