open-mmlab/OpenPCDet #1690: Keyerror: 'bbox'

Chenycan, 2024-12-05:
Traceback (most recent call last):
  File "tools/train.py", line 233, in <module>
    main()
  File "tools/train.py", line 223, in main
    repeat_eval_ckpt(
  File "/home/guest/2TLOCAL/CYC/pointpillars_test/OpenPCDet/tools/test.py", line 123, in repeat_eval_ckpt
    tb_dict = eval_utils.eval_one_epoch(
  File "/home/guest/2TLOCAL/CYC/pointpillars_test/OpenPCDet/tools/eval_utils/eval_utils.py", line 125, in eval_one_epoch
    result_str, result_dict = dataset.evaluation(
  File "/home/guest/2TLOCAL/CYC/pointpillars_test/OpenPCDet/pcdet/datasets/custom/custom_dataset.py", line 337, in evaluation
    ap_result_str, ap_dict = kitti_eval.get_official_eval_result(eval_gt_annos, eval_det_annos, class_names)
  File "/home/guest/2TLOCAL/CYC/pointpillars_test/OpenPCDet/pcdet/datasets/custom/kitti_object_eval_python/eval.py", line 673, in get_official_eval_result
    mAPbbox, mAPbev, mAP3d, mAPaos, mAPbbox_R40, mAPbev_R40, mAP3d_R40, mAPaos_R40 = do_eval(
  File "/home/guest/2TLOCAL/CYC/pointpillars_test/OpenPCDet/pcdet/datasets/custom/kitti_object_eval_python/eval.py", line 589, in do_eval
    ret = eval_class(gt_annos, dt_annos, current_classes, difficultys, 0,
  File "/home/guest/2TLOCAL/CYC/pointpillars_test/OpenPCDet/pcdet/datasets/custom/kitti_object_eval_python/eval.py", line 475, in eval_class
    rets = calculate_iou_partly(dt_annos, gt_annos, metric, num_parts)
  File "/home/guest/2TLOCAL/CYC/pointpillars_test/OpenPCDet/pcdet/datasets/custom/kitti_object_eval_python/eval.py", line 361, in calculate_iou_partly
    gt_boxes = np.concatenate([a["bbox"] for a in gt_annos_part], 0)
  File "/home/guest/2TLOCAL/CYC/pointpillars_test/OpenPCDet/pcdet/datasets/custom/kitti_object_eval_python/eval.py", line 361, in <listcomp>
    gt_boxes = np.concatenate([a["bbox"] for a in gt_annos_part], 0)
KeyError: 'bbox'

When i try to train my custom dataset using kitti as a template, i encounter this question. Can anyone solve this problem?

Chenycan, 2024-12-09:
please somebody help me!

Petros626, 2024-12-09:
@Chenycan which script you execute? you must provide more details.

closed 2025-01-04 by Chenycan

cbcabc, 2025-07-15:
> It's missing 2d bounding boxes for evaluation. Your gt or prediction labels aren't in the required kitti format for evaluation.
Thanks for your replying. But I remember that some codes which containing "bbox" have been commented in custom_dataset.py and .yaml file. However, it's still not working.

Petros626, 2025-07-15:
I can just speak for the KITTI dataset. There is a required standard format for the evaluation. I didn't worked with custom_dataset.py, but refer to the kitti_dataset.py and implement the code you need to perform evaluation.