open-mmlab/OpenPCDet #1692: KITTI eval does not count FNs?

Kh-Hamed, 2024-12-09:
Hi,

I saw the KITTI eval metric. In eval.py there is a function called do_eval. The eval_class returns a dictionary, which contains precision, recall, and aos. The problem is that after that, instead of calculating the area under the P_R curve, the function only passes ret["precision"] for the calculation final result. Is this Wrong? It does not count any of the FNs in this way. It does not use the recall values!!!
Here is the code I'm talking about:

https://github.com/open-mmlab/OpenPCDet/blob/master/pcdet/datasets/kitti/kitti_object_eval_python/eval.py#L590
[image]
[image]

Kh-Hamed, 2025-01-03:
Hi friends,
Does anybody have any idea? This is important to me.

Petros626, 2025-01-08:
The repo seems not to be maintained anymore. I would debug the code via Jupyter Notebook and see, if your assumptions are wrong or not. When nobody mentioned anything in all the issues, you have to solve it yourself.

closed 2025-02-22 by github-actions[bot]