open-mmlab/OpenPCDet #1695: object detected but dimension seem not good

thanhpt93, 2024-12-24:
Hi author, I have issue after I train and test my model, object was detected but dimension seem not good, maybe my anchor size is fail? Can you help me edit some config to get good result?
Bellow is result image:
[image]

For more detail: I train with custom dataset, base pv_rcnn.yaml config and have 3 class
Thank you very much!

Petros626, 2024-12-31:
I remember to got this behaviour the first run, I used the `demo.py` code. Search in the `O3DVisualizer` where the boxes are drawn.

thanhpt93, 2024-12-31:
@Petros626  Thank you, I found reason

Petros626, 2024-12-31:
@thanhpt93 please share the solution for other users and close then the issue, if no further debugging is necessary.

thanhpt93, 2024-12-31:
@Petros626 Thank you, it was simply me handling the data custom incorrectly, swapping the height and length.

closed 2025-01-03 by thanhpt93