open-mmlab/OpenPCDet #1701: Encountered a ninja error while installing pcdet.

HongyuanLI17, 2025-01-12:
Hi, guys. I encountered an error with ninja while installing pcdet according to the install.md. How can I get the build.ninja file?
 Here are the versions of the key software I'm using:
CUDA:11.3
PyTorch：1.10.0
GCC:9.5
ninja:1.10.2
re2c:3.0
I'd really appreciate it if anyone could offer some insights or suggestions based on this information. Thanks in advance!

[image]

Petros626, 2025-01-12:
I think, I posted in the issues a installation guide.

HongyuanLI17, 2025-01-13:
I just wanted to say thank you for your help.I've managed to solve the problem. I reinstalled CUDA and ninja, upgraded torch to 12.1，and also installed torch-scatter before running setup.py.And now it works! Thanks again！

closed 2025-01-13 by HongyuanLI17

WilliamFlinchbaugh, 2025-01-20:
I'm having a very similar issue, how did you fix this? I'm using a conda environment with python 3.8. I'm on CUDA 12.3, but I'm using the most recent torch with CUDA 12.1, and spconv-cu120. 

What did you do differently to get it working with CUDA 12.1?

Here's exactly what I did: 
`conda create --name openpcdet python=3.8`
`conda activate openpcdet`
`pip3 install torch torchvision torchaudio --index-url https://download.pytorch.org/whl/cu121`
`pip3 install spconv-cu120`
`pip3 install -r requirements.txt`
`python setup.py develop`

Any help would be greatly appreciated!

Petros626, 2025-04-05:
@WilliamFlinchbaugh still struggling?