open-mmlab/OpenPCDet #1704: can you share me the pre-train model

superkissman, 2025-02-12:
> for me all links of the pretrained nuscenes models are working.> 
> > 
> can you share me the working link for [CenterPoint (voxel_size=0.1)](https://github.com/open-mmlab/OpenPCDet/blob/master/tools/cfgs/nuscenes_models/cbgs_voxel01_res3d_centerpoint.yaml) model? Thanks! 

 _Originally posted by @hongbozheng in [#1673](https://github.com/open-mmlab/OpenPCDet/issues/1673#issuecomment-2429246659)_

hongbozheng, 2025-02-12:
I think both [CenterPoint (voxel_size=0.075)](https://github.com/open-mmlab/OpenPCDet/blob/master/tools/cfgs/nuscenes_models/cbgs_voxel0075_res3d_centerpoint.yaml) and [CenterPoint (voxel_size=0.1)](https://github.com/open-mmlab/OpenPCDet/blob/master/tools/cfgs/nuscenes_models/cbgs_voxel01_res3d_centerpoint.yaml) pre-trained models are not available right now. I did get a chance to download [CenterPoint (voxel_size=0.075)](https://github.com/open-mmlab/OpenPCDet/blob/master/tools/cfgs/nuscenes_models/cbgs_voxel0075_res3d_centerpoint.yaml) a while ago when it's still available, but [CenterPoint (voxel_size=0.1)](https://github.com/open-mmlab/OpenPCDet/blob/master/tools/cfgs/nuscenes_models/cbgs_voxel01_res3d_centerpoint.yaml) is never available for me.

hongbozheng, 2025-02-17:
@superkissman Do you want me to share it with you? Does a Google Drive link work for you?

hongbozheng, 2025-02-17:
@superkissman You can download the pre-trained weights of [CenterPoint (voxel_size=0.075)](https://github.com/open-mmlab/OpenPCDet/blob/master/tools/cfgs/nuscenes_models/cbgs_voxel0075_res3d_centerpoint.yaml) [here](https://drive.google.com/file/d/1Ltr3wb71dlXVqYO2MmPtrjKiRg7Ub_pT/view?usp=sharing).

superkissman, 2025-02-18:
cool! Thanks for sharing.

Petros626, 2025-03-22:
@hongbozheng other centernet aren't available or we're downloaded by you or others?

hongbozheng, 2025-03-22:
@Petros626 Unfortunately, I only have access to [CenterPoint (voxel_size=0.075)](https://github.com/open-mmlab/OpenPCDet/blob/master/tools/cfgs/nuscenes_models/cbgs_voxel0075_res3d_centerpoint.yaml) [here](https://drive.google.com/file/d/1Ltr3wb71dlXVqYO2MmPtrjKiRg7Ub_pT/view?usp=sharing). I did not get a chance to download others as they're never available for me.

Petros626, 2025-03-22:
https://u-173-c219.cs.uni-tuebingen.de/zhan/mmdetection3d/-/tree/test-1.2/configs/Centerpoint this could help,. 0.1 and 0.075 voxel size available.

closed 2025-05-07 by github-actions[bot]

SharangKaul123, 2026-02-24:
@hongbozheng Do you still have the [CenterPoint (voxel_size=0.075)](https://github.com/open-mmlab/OpenPCDet/blob/master/tools/cfgs/nuscenes_models/cbgs_voxel0075_res3d_centerpoint.yaml) checkpoint? The google drive link does not work at all.

SharangKaul123, 2026-02-24:
@hongbozheng  Thank you so much. It worked now :)