open-mmlab/OpenPCDet #1705: I need Assitance with how the workflow of this codebase  if I wanna use my own 3d LIDAR dataset

Mo7amedNajjar, 2025-02-13:
Hello I have My own 3d lidar custom dataset which includes npy files (Raw point cloud files) collected with velodyne 128 CHannel LIDAR and corresponding labels in txt files. My objective is to detect the human in the the scene and be able to train and test and get validation scores.

I am really not sure where to start, I am not very good with software so it is complicated to me to understand which files in OPENPCDET codebase that I would use to complete this task. 

If someone is willing to help to go through this thorouly I would really appreciate it, and would consider paying you if you could help me understand and Implement the task I am trying to achieve.  Shoot me a message if you are willing to help me out.

llz666ha, 2025-02-17:
You can refer to some blogs,such as: https://blog.csdn.net/zyw2002/article/details/127419245  .
search on the Internet

Petros626, 2025-02-21:
First of all you should ensure that your data matches the unified coordinate system of `OpenPCDet design pattern`. Some users created a tutorial for using a custom dataset here: https://github.com/open-mmlab/OpenPCDet/blob/master/docs/CUSTOM_DATASET_TUTORIAL.md. The issues are full of topics about the usage of the custom dataset and it's adjustments for `OpenPCDet`. 

Further you can take a look here to understand this toolbox better:

[OpenPCDet_Tutorial.pdf](https://github.com/user-attachments/files/18904780/OpenPCDet_Tutorial.pdf)

I would first ensure, that the data from your lidar equals to a know format like `KITTI`, so that you can use the default scripts for the related dataset. Then visualize several scenes with gt boxes to make youre everything is correct. Then you can prepare your training job with the `.yaml` file and apply your specific needs (`pc_range`, augmentations, etc.). It's sufficient that you have enough data for training to perform well.

Petros626, 2025-02-21:
> You can refer to some blogs,such as: https://blog.csdn.net/ZhuanShangNiDeXin/article/details/131132040 . search on the Internet

https://github.com/open-mmlab/OpenPCDet/issues/1680

closed 2025-04-07 by github-actions[bot]