open-mmlab/OpenPCDet #1706: The relationship between POINT_CLOUD_RANGE and VOXEL_SIZE

llz666ha, 2025-02-17:
What is the relationship between POINT_CLOUD_RANGE and VOXEL_SIZE? What are the restrictions? In addition to the "Point cloud range along z-axis/voxel_size is 40. Point cloud range along x&y-axis / voxel_size is the multiple of 16.”mentioned by the author.
What part of the relevant code?Please! I need learn it.

llz666ha, 2025-02-21:
Thank you for your reply. In fact, I used two projects built on the basis of Openpcdet: [CasA](https://github.com/hailanyi/CasA) and [TED](https://github.com/hailanyi/TED/tree/master) The requirements of these two projects for point_cloud_range and voxel_size seem to have changed, such as :z-axis/voxel_size is 80.  So I want to know what is the restriction of the relationship between these two attributes

Petros626, 2025-02-21:
relation: The point cloud range along each axis (x, y, z) is divided by the corresponding voxel size to determine the number of voxels along that axis.

restrictions: The number of voxels along the z-axis is less strict but should still be chosen carefully

The voxelization process is typically handled by the `VoxelGenerator` class in `OpenPCDet`. Look for the `transform_points_to_voxels` function in the codebase.