open-mmlab/OpenPCDet #1710: Applying the format of the kitti dataset to bring in our own data but results have an error in test——N > 0 assert faild. CUDA kernel launch blocks must be positive, but got N= 0

Thehent, 2025-02-24:
as the title said and this is my logging, model is pv_rcnn, maybe it's my data's fault.
(openpcdet) apple@ubuntu:~/InScope/OpenPCDet-master/tools$ python "/home/apple/InScope/OpenPCDet-master/tools/test.py" --cfg_file "/home/apple/InScope/OpenPCDet-master/tools/cfgs/kitti_models/pv_rcnn.yaml" --batch_size=1 --ckpt "/home/apple/InScope/OpenPCDet-master/output/home/apple/InScope/OpenPCDet-master/tools/cfgs/kitti_models/pv_rcnn/default/ckpt/checkpoint_epoch_1.pth"
2025-02-24 16:04:06,968   INFO  **********************Start logging**********************
2025-02-24 16:04:06,968   INFO  CUDA_VISIBLE_DEVICES=ALL
2025-02-24 16:04:06,968   INFO  cfg_file         /home/apple/InScope/OpenPCDet-master/tools/cfgs/kitti_models/pv_rcnn.yaml
2025-02-24 16:04:06,968   INFO  batch_size       1
2025-02-24 16:04:06,968   INFO  workers          4
2025-02-24 16:04:06,968   INFO  extra_tag        default
2025-02-24 16:04:06,968   INFO  ckpt             /home/apple/InScope/OpenPCDet-master/output/home/apple/InScope/OpenPCDet-master/tools/cfgs/kitti_models/pv_rcnn/default/ckpt/checkpoint_epoch_1.pth
2025-02-24 16:04:06,968   INFO  pretrained_model None
2025-02-24 16:04:06,968   INFO  launcher         none
2025-02-24 16:04:06,968   INFO  tcp_port         18888
2025-02-24 16:04:06,968   INFO  local_rank       None
2025-02-24 16:04:06,968   INFO  set_cfgs         None
2025-02-24 16:04:06,968   INFO  max_waiting_mins 30
2025-02-24 16:04:06,968   INFO  start_epoch      0
2025-02-24 16:04:06,968   INFO  eval_tag         default
2025-02-24 16:04:06,968   INFO  eval_all         False
2025-02-24 16:04:06,968   INFO  ckpt_dir         None
2025-02-24 16:04:06,968   INFO  save_to_file     False
2025-02-24 16:04:06,968   INFO  infer_time       False
2025-02-24 16:04:06,968   INFO  cfg.ROOT_DIR: /home/apple/InScope/OpenPCDet-master
2025-02-24 16:04:06,968   INFO  cfg.LOCAL_RANK: 0
2025-02-24 16:04:06,968   INFO  cfg.CLASS_NAMES: ['Car']
2025-02-24 16:04:06,968   INFO  ----------- DATA_CONFIG -----------
2025-02-24 16:04:06,968   INFO  cfg.DATA_CONFIG.DATASET: KittiDataset
2025-02-24 16:04:06,968   INFO  cfg.DATA_CONFIG.DATA_PATH: ../data/kitti
2025-02-24 16:04:06,968   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [0, -40, -3, 70.4, 40, 1]
2025-02-24 16:04:06,968   INFO  ----------- DATA_SPLIT -----------
2025-02-24 16:04:06,968   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: train
2025-02-24 16:04:06,968   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: val
2025-02-24 16:04:06,968   INFO  ----------- INFO_PATH -----------
2025-02-24 16:04:06,968   INFO  cfg.DATA_CONFIG.INFO_PATH.train: ['kitti_infos_train.pkl']
2025-02-24 16:04:06,968   INFO  cfg.DATA_CONFIG.INFO_PATH.test: ['kitti_infos_val.pkl']
2025-02-24 16:04:06,968   INFO  cfg.DATA_CONFIG.GET_ITEM_LIST: ['points']
2025-02-24 16:04:06,968   INFO  cfg.DATA_CONFIG.FOV_POINTS_ONLY: True
2025-02-24 16:04:06,968   INFO  ----------- DATA_AUGMENTOR -----------
2025-02-24 16:04:06,968   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder']
2025-02-24 16:04:06,968   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'gt_sampling', 'USE_ROAD_PLANE': False, 'DB_INFO_PATH': ['kitti_dbinfos_train.pkl'], 'PREPARE': {'filter_by_min_points': ['Car:1'], 'filter_by_difficulty': [-1]}, 'SAMPLE_GROUPS': ['Car:15'], 'NUM_POINT_FEATURES': 4, 'DATABASE_WITH_FAKELIDAR': False, 'REMOVE_EXTRA_WIDTH': [0.0, 0.0, 0.0], 'LIMIT_WHOLE_SCENE': False}, {'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.78539816, 0.78539816]}, {'NAME': 'random_world_scaling', 'WORLD_SCALE_RANGE': [0.95, 1.05]}]
2025-02-24 16:04:06,968   INFO  ----------- POINT_FEATURE_ENCODING -----------
2025-02-24 16:04:06,968   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
2025-02-24 16:04:06,968   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.used_feature_list: ['x', 'y', 'z', 'intensity']
2025-02-24 16:04:06,968   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.src_feature_list: ['x', 'y', 'z', 'intensity']
2025-02-24 16:04:06,968   INFO  cfg.DATA_CONFIG.DATA_PROCESSOR: [{'NAME': 'mask_points_and_boxes_outside_range', 'REMOVE_OUTSIDE_BOXES': True}, {'NAME': 'shuffle_points', 'SHUFFLE_ENABLED': {'train': True, 'test': False}}, {'NAME': 'transform_points_to_voxels', 'VOXEL_SIZE': [0.05, 0.05, 0.1], 'MAX_POINTS_PER_VOXEL': 5, 'MAX_NUMBER_OF_VOXELS': {'train': 16000, 'test': 40000}}]
2025-02-24 16:04:06,968   INFO  cfg.DATA_CONFIG._BASE_CONFIG_: /home/apple/InScope/OpenPCDet-master/tools/cfgs/dataset_configs/kitti_dataset.yaml
2025-02-24 16:04:06,968   INFO  ----------- MODEL -----------
2025-02-24 16:04:06,968   INFO  cfg.MODEL.NAME: PVRCNN
2025-02-24 16:04:06,968   INFO  ----------- VFE -----------
2025-02-24 16:04:06,968   INFO  cfg.MODEL.VFE.NAME: MeanVFE
2025-02-24 16:04:06,968   INFO  ----------- BACKBONE_3D -----------
2025-02-24 16:04:06,968   INFO  cfg.MODEL.BACKBONE_3D.NAME: VoxelBackBone8x
2025-02-24 16:04:06,968   INFO  ----------- MAP_TO_BEV -----------
2025-02-24 16:04:06,968   INFO  cfg.MODEL.MAP_TO_BEV.NAME: HeightCompression
2025-02-24 16:04:06,968   INFO  cfg.MODEL.MAP_TO_BEV.NUM_BEV_FEATURES: 256
2025-02-24 16:04:06,968   INFO  ----------- BACKBONE_2D -----------
2025-02-24 16:04:06,969   INFO  cfg.MODEL.BACKBONE_2D.NAME: BaseBEVBackbone
2025-02-24 16:04:06,969   INFO  cfg.MODEL.BACKBONE_2D.LAYER_NUMS: [5, 5]
2025-02-24 16:04:06,969   INFO  cfg.MODEL.BACKBONE_2D.LAYER_STRIDES: [1, 2]
2025-02-24 16:04:06,969   INFO  cfg.MODEL.BACKBONE_2D.NUM_FILTERS: [128, 256]
2025-02-24 16:04:06,969   INFO  cfg.MODEL.BACKBONE_2D.UPSAMPLE_STRIDES: [1, 2]
2025-02-24 16:04:06,969   INFO  cfg.MODEL.BACKBONE_2D.NUM_UPSAMPLE_FILTERS: [256, 256]
2025-02-24 16:04:06,969   INFO  ----------- DENSE_HEAD -----------
2025-02-24 16:04:06,969   INFO  cfg.MODEL.DENSE_HEAD.NAME: AnchorHeadSingle
2025-02-24 16:04:06,969   INFO  cfg.MODEL.DENSE_HEAD.CLASS_AGNOSTIC: False
2025-02-24 16:04:06,969   INFO  cfg.MODEL.DENSE_HEAD.USE_DIRECTION_CLASSIFIER: True
2025-02-24 16:04:06,969   INFO  cfg.MODEL.DENSE_HEAD.DIR_OFFSET: 0.78539
2025-02-24 16:04:06,969   INFO  cfg.MODEL.DENSE_HEAD.DIR_LIMIT_OFFSET: 0.0
2025-02-24 16:04:06,969   INFO  cfg.MODEL.DENSE_HEAD.NUM_DIR_BINS: 2
2025-02-24 16:04:06,969   INFO  cfg.MODEL.DENSE_HEAD.ANCHOR_GENERATOR_CONFIG: [{'class_name': 'Car', 'anchor_sizes': [[3.9, 1.6, 1.56]], 'anchor_rotations': [0, 1.57], 'anchor_bottom_heights': [-1.78], 'align_center': False, 'feature_map_stride': 8, 'matched_threshold': 0.6, 'unmatched_threshold': 0.45}]
2025-02-24 16:04:06,969   INFO  ----------- TARGET_ASSIGNER_CONFIG -----------
[... 154 lines cut ...]
    pred_dicts, ret_dict = model(batch_dict)
  File "/home/apple/data2/anaconda3/envs/openpcdet/lib/python3.8/site-packages/torch/nn/modules/module.py", line 889, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/home/apple/InScope/OpenPCDet-master/tools/../pcdet/models/detectors/pv_rcnn.py", line 11, in forward
    batch_dict = cur_module(batch_dict)
  File "/home/apple/data2/anaconda3/envs/openpcdet/lib/python3.8/site-packages/torch/nn/modules/module.py", line 889, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/home/apple/InScope/OpenPCDet-master/tools/../pcdet/models/backbones_3d/spconv_backbone.py", line 149, in forward
    x = self.conv_input(input_sp_tensor)
  File "/home/apple/data2/anaconda3/envs/openpcdet/lib/python3.8/site-packages/torch/nn/modules/module.py", line 889, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/home/apple/data2/anaconda3/envs/openpcdet/lib/python3.8/site-packages/spconv/pytorch/modules.py", line 138, in forward
    input = module(input)
  File "/home/apple/data2/anaconda3/envs/openpcdet/lib/python3.8/site-packages/torch/nn/modules/module.py", line 889, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/home/apple/data2/anaconda3/envs/openpcdet/lib/python3.8/site-packages/spconv/pytorch/conv.py", line 755, in forward
    return self._conv_forward(self.training,
  File "/home/apple/data2/anaconda3/envs/openpcdet/lib/python3.8/site-packages/spconv/pytorch/conv.py", line 408, in _conv_forward
    raise e
  File "/home/apple/data2/anaconda3/envs/openpcdet/lib/python3.8/site-packages/spconv/pytorch/conv.py", line 385, in _conv_forward
    res = ops.get_indice_pairs_implicit_gemm(
  File "/home/apple/data2/anaconda3/envs/openpcdet/lib/python3.8/site-packages/spconv/pytorch/ops.py", line 521, in get_indice_pairs_implicit_gemm
    SpconvOps.generate_subm_conv_inds(inds_tv,
RuntimeError: /tmp/pip-build-env-g7ks_t5y/overlay/lib/python3.8/site-packages/cumm/include/tensorview/cuda/launch.h(54)
N > 0 assert faild. CUDA kernel launch blocks must be positive, but got N= 0

Hberto, 2025-02-27:
any new updates in this? I have the same issue when executing test.py

Petros626, 2025-02-27:
Isn't this more a Cuda issue? I never experienced that.

closed 2025-04-13 by github-actions[bot]