open-mmlab/OpenPCDet #1711: Issue when starting training my own custom dataset

Mo7amedNajjar, 2025-02-26:
when I run the  command  python tools/train.py --cfg_file tools/cfgs/custom_models/second.yaml --batch_size 4   

I get this error :

[image]

So I am basically following the OPENPCDET custom dataset tutorial, I followed everything from there. 

But like is there any specifc file that I need to configure in order to get the training working? 

What are things do I need to consider looking at ?

Petros626, 2025-02-26:
Did you make sure, that your data are ok, like i mentioned before? Did you visualized them?

Petros626, 2025-02-26:
https://github.com/open-mmlab/OpenPCDet/issues/1162, https://github.com/open-mmlab/OpenPCDet/issues/253#issuecomment-679190936

Mo7amedNajjar, 2025-02-26:
Hey Petros,  Sorry so once I have my raw point clouds in ( points) in npy format.   and my one to one matching labels in (labels) in txt format. 

What are the neccessary steps to make sure that my data is ok ?  Like when you say to visualzie them, how can I do that ?  Is there a script in OPENPCDET to check that ? 

I am sorry if my questions are very vague but I am honestly just confused and tried to understand the workflow of the project but my software knowledge is very limited.

pangzi-debug, 2025-04-11:
Seems it is a issue with voxel size in cusom_dataset.yaml，Maybe the network need well-tailed voxelized scene for its training. In my case, I change the voxel size from [[0.1,0.1,0.15]] to [[0.05,0.05,0.1]],then the code works.

pangzi-debug, 2025-04-12:
> Seems it is a issue with voxel size in cusom_dataset.yaml，Maybe the network need well-tailed voxelized scene for its training. In my case, I change the voxel size from [[0.1,0.1,0.15]] to [[0.05,0.05,0.1]],then the code works.

May be you haven't abey the rule:
'''For voxel based detectors such as SECOND, PV-RCNN and CenterPoint, the point cloud range and voxel size should follow:
1. Point cloud range along z-axis / voxel_size is 40
2. Point cloud range along x&y-axis / voxel_size is the multiple of 16.'''