open-mmlab/OpenPCDet #1718: OpenPCDET detection implementation

Mo7amedNajjar, 2025-03-24:
I was wondering if Openpcdet could classify detections considering more than one frame. For example I have data collected from VLP 128 channel data which contains a human performing activities like walking, jumping, squatting. 

Could the network be configured to take in 5 frames for example to make a detection ? Instead of just taking frame by frame ? 

Sorry if my question does not make sense, but what would I have to do interms of setting up the data and network to make something like that possible ?

cheng052, 2025-04-02:
You may check this, MPPNet: Multi-Frame Feature Intertwining with Proxy Points for 3D Temporal Object Detection
https://arxiv.org/pdf/2205.05979
https://github.com/open-mmlab/OpenPCDet/blob/master/docs/guidelines_of_approaches/mppnet.md

Petros626, 2025-04-03:
You mean like a tracking approach? You can train a openpcdet model on a dataset and then use sequences from the dataset for predictions in sequence.

Qu0rise, 2025-04-15:
Hi @Mo7amedNajjar ,

Good question about classifying actions like walking/jumping/squatting over multiple frames (e.g., 5 frames) using OpenPCDet.

Currently, OpenPCDet focuses on per-frame detection and tracking, not directly classifying longer-term actions based on sequences. While models like MPPNet (@cheng052's suggestion) use multi-frame info to *improve detection*, they aren't designed for high-level action classification out-of-the-box.

The standard way to tackle this is usually a **two-stage approach**:

1.  **Detect & Track:** Use OpenPCDet for robust person detection on your VLP-128 data, then track them across frames (as @Petros626 alluded to) to get consistent IDs.
2.  **Classify Actions:** For each tracked person, extract temporal features (e.g., how their 3D bounding box moves, changes in point cloud shape/density over the 5 frames). Feed these feature sequences into a separate classifier (LSTM, Transformer, etc.) trained to recognize 'walking', 'jumping', 'squatting'.

So, rather than modifying an OpenPCDet detector to directly take 5 frames for action classification (which would be complex), the more practical path is: **OpenPCDet (Detect/Track) -> Feature Extractor -> Temporal Classifier (LSTM/Transformer)**.

You'll need sequence data with action labels to train that final classification step.

Hope this helps clarify a potential direction!

Petros626, 2025-04-15:
@Mo7amedNajjar this could solve your problem: https://github.com/YCAyca/3D-Multi-Object-Tracking-using-Lidar, https://www.kitware.com/3d-multi-object-tracking-using-lidar-for-autonomous-driving/. 

Yagmur trained a 3D detector with `OpenPCDet` and used a Kalman Filter to Track KITTI sequences (guess this here: https://www.cvlibs.net/datasets/kitti/eval_tracking.php).

closed 2025-05-31 by github-actions[bot]