open-mmlab/OpenPCDet #1735: The reproduction accuracy has decreased.

liying715, 2025-07-09:
Why is the accuracy of the latest project reproduction much lower than that provided on the official website? Could you please provide the versions of the required dependency packages for reproduction?Because I have reproduced it before.

closed 2025-08-24 by github-actions[bot]

AmirKaramiFini, 2025-11-01:
@Petros626 do you know anything about this ? im trying to reproduce any model base on pretrain weights but i cant achieve what mentioned in papers. i tried with many dependencies even the docker image but im failing.

Petros626, 2025-11-01:
> [@Petros626](https://github.com/Petros626) do you know anything about this ? im trying to reproduce any model base on pretrain weights but i cant achieve what mentioned in papers. i tried with many dependencies even the docker image but im failing.

I never reproduced any 3D model of this codebase and compare it to a paper. Which paper? 
I used venv installation in my case, but an adapted docker file to your package versions should work.

AmirKaramiFini, 2025-11-01:
> I never reproduced any 3D model of this codebase and compare it to a paper. Which paper? I used venv installation in my case, but an adapted docker file to your package versions should work.

in papers like [CasA](https://github.com/hailanyi/CasA) or [SFD ](https://github.com/LittlePey/SFD), for example they are reporting the AP of PV-RCNN with R40 these numbers on validation set : 
92.57 84.83 82.69 for easy , moderate and hard respectively

but when im running the provided weight from model zoo and the config, im getting this result : 

[image]

at first i thought it was my dependency problem, so i tried to run with[ this docker image](https://hub.docker.com/layers/djiajun1206/pcdet/python3.8_pytorch1.10/images/sha256-8042f89e917da9b292685fb951fb1aa04ac9033d2319d35e726012113b02f953) which is provided before. but the result was the same. 

the problem is the same for other model provided in the [model zoo](https://github.com/open-mmlab/OpenPCDet?tab=readme-ov-file#model-zoo) and the gap between reported results in the papers(all of them) and my results are even larger than the numbers i mentioned before.

i also tried with 0.5.2 and 0.5.0 version of pcdet but the result was the same
am i missing something ?

Petros626, 2025-11-01:
You need the exact same configuration of the OpenMMLab Team or from the paper, to achieve approx. their results. Make sure that the 40 point interpolation is also implemented for OpenPCDet, when I remember correctly, they use both, which is okay.

Why it should be an dependency problem? You train the models weights, doesn't matter if as venv or Docker, the results stay the same. When you get better results, than in the paper be satisfied with it, only worse results should be questioned and investigated.

Petros626, 2025-11-01:
First you posted an image showing better results than mentioned in the text and then worse results. I'm confused.

Nevermind your results are near the paper results, as I said you need to know, which configuration does OpenMMLab or the paper authors used to achieve these results. Try the OpenPCDet conf and if provided the conf of the paper and compare hyperparameters of the model and dataset configuration. Maybe then you will be on the same level. Data augmentation plays also a crucial role.